basler/pypylon #159: Enumeration of camera - issue with multiple instances of the same camera!

harshakasi, 2019-10-14:
I am trying to use the USB 3.0 1280-54um camera through pypylon on my Mac OX 10.15 (Catalina). This error has nothing to do with the Mac OSx latest update from 10.14 (Mojave) to 10.15 as the error was pre-existing in the 10.14 revision.

Based on the examples for camera access and functioning provided in the python code examples, I am doing the following in my _`__main__`_:

`
self.tlFactory = pylon.TlFactory.GetInstance()                    # instantiate an object
self.devList = self.tlFactory.EnumerateDevices()               # enumerate devices 
`

At this point, although there is only one Basler camera 1280-54um connected to the USB port, the _devList_ object is assigned a list of two devices both containing the same serial number.

A snapshot of the content within _devList_ as obtained from Visual Code Editor is:
_0:<pypylon.pylon.DeviceInfo; proxy of <Swig Object of type 'Pylon::CDeviceInfo *' at 0x118614f30> >
1:<pypylon.pylon.DeviceInfo; proxy of <Swig Object of type 'Pylon::CDeviceInfo *' at 0x118614ea0> >
__len__:2_

@basler-oss, could you please provide a resolution for the same at the earliest?

I am trying to make my app work on python and at the tranport layer somehow two instances of the same device get created within the EnumerateDevices() function.

thiesmoeller, 2019-10-16:
if you print out the device-info [ e.g. model and serial number ] of the two devices you see:
are they really the same? 
As a quick workaround the method CreateFirstDevice() on TlFactory will pick the first on any list

thiesmoeller, 2020-12-02:
is the issue still present with most current release?

harshakasi, 2021-05-07:
> is the issue still present with the most current release?

Yes it is still present in the current release Version 1.7.2
and probably why in the sample code (startup.py) snippet, they suggest using the following instruction to create a camera object pertaining to the connected device:

camera = pylon.InstantCamera(
        pylon.TlFactory.GetInstance().CreateFirstDevice())

Hope that answers your query.

Thanks!

harshakasi, 2021-05-07:
> if you print out the device-info [ e.g. model and serial number ] of the two devices you see:
> are they really the same?
> As a quick workaround the method CreateFirstDevice() on TlFactory will pick the first on any list

That's exactly what works too, not a workaround but the only way I guess to ensure first of the device objects get used.

thiesmoeller, 2021-05-10:
@harshakasi 

can you please post the the output of the following script, on the system that shows this issue:

```python
import pypylon.pylon as py

tlf = py.TlFactory.GetInstance()

for tl in tlf.EnumerateTls():
    print(tl.GetDeviceClass(), tl.GetFileName(), tl.GetFullName())

```

harshakasi, 2021-05-17:
@thiesmoeller 

Sorry for the delay in my response. Here is the output for the code snippet, Thies:

```
BaslerUsb N/A USB/BaslerUsb 6.1.2.18349
BaslerUsb N/A USB/BaslerUsb 6.1.2.18349
BaslerGigE N/A GigE/BaslerGigE 6.1.2.18349
BaslerGigE N/A GigE/BaslerGigE 6.1.2.18349
```

Please advise.

Best,

Harsha

thiesmoeller, 2021-05-17:
the copying of the files into the wheel has an issue of accidentally resolving symbolic links.
pylon tries to enumerate all transport layers and will find the same library under two names .

```
mac@macs-MacBook-Pro ~ % sudo dtrace -n 'pid$target:libdyld:dlopen:entry { printf("dlopen: %s\n", copyinstr(arg0));}  ' -c "/Users/mac/.pyenv/versions/3.7.9/bin/python3 test.py"
dtrace: system integrity protection is on, some features will not be available

dtrace: description 'pid$target:libdyld:dlopen:entry ' matched 1 probe
BaslerUsb N/A USB/BaslerUsb 6.1.2.18349
BaslerGigE N/A GigE/BaslerGigE 6.1.2.18349
BaslerGigE N/A GigE/BaslerGigE 6.1.2.18349
dtrace: pid 5883 has exited
CPU     ID                    FUNCTION:NAME
  1   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/_pylon.cpython-37m-darwin.so

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/pylon.framework/Versions/A/Libraries/liblog4cpp_gcc_v3_1_Basler_pylon.dylib

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/_genicam.cpython-37m-darwin.so

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/lib-dynload/_heapq.cpython-37m-darwin.so

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/pylon.framework/Versions/A/Libraries/libpylon_TL_camemu-6.1.2.so

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/pylon.framework/Versions/A/Libraries/libpylon_TL_camemu.so

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/pylon.framework/Versions/A/Libraries/libpylon_TL_gige-6.1.2.so

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/pylon.framework/Versions/A/Libraries/libpylon_TL_gige.so

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/pylon.framework/Versions/A/Libraries/libpylon_TL_gtc-6.1.2.so

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/pylon.framework/Versions/A/Libraries/libpylon_TL_gtc.so

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/pylon.framework/Versions/A/Libraries/libpylon_TL_usb-6.1.2.so

  3   2358                     dlopen:entry dlopen: /Users/mac/.pyenv/versions/3.7.9/lib/python3.7/site-packages/pypylon/pylon.framework/Versions/A/Libraries/pylon-libusb-1.0.dylib

```

harshakasi, 2021-05-21:
Wow! Thanks.
So what should be my resolution for this, please? I would like the output for enumerate with just the connected devices and not duplicates of them.

thiesmoeller, 2021-10-20:
check this ticket also:

https://github.com/basler/pypylon/issues/424#issuecomment-947522532

thiesmoeller, 2025-12-15:
Closing as answered. The duplicate enumeration issue is caused by symbolic links being resolved during wheel packaging. As a workaround, use `CreateFirstDevice()` to get the first device from the enumeration list. This is the recommended approach when you have a single camera. The underlying issue is a known limitation in how transport layers are packaged. See also issue #424 for related discussion.