basler/pypylon #160: Camera stops working after some time

lusaeg, 2019-10-17:
Hi there,
I'm programming a GUI, where live images of multiple cameras are displayed simultanously. This is a simplified snippet from the code I'm using:
```
class baslercam:
  def __init__(self):
    tlFactory = pylon.TlFactory.GetInstance()
    self.devices = tlFactory.EnumerateDevices()
    if len(self.devices) == 0:
        raise pylon.RUNTIME_EXCEPTION("No camera present.")
    
    self.cameras = pylon.InstantCameraArray(len(self.devices))
    
    for i, cam in enumerate(self.cameras):
        cam.Attach(tlFactory.CreateDevice(self.devices[i]))
    
    self.numberOfFramesGrabbed = 0
    self.allCamerasWorking = True

 def grabImageToArray(self,order):
    self.cameras.StartGrabbing()
    for i in range(len(self.devices)):
      if not self.cameras.IsGrabbing():
        break
      grabResult = self.cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
      if grabResult.GrabSucceeded():
        print('Grab successful')
      else:
        print('Grab not successful')
      cameraContextValue = grabResult.GetCameraContext()
      print('Camera Context: '+str(cameraContextValue))
      converter = pylon.ImageFormatConverter()
      image0 = converter.Convert(grabResult)
      if i == 0:
        img0 = [image0.GetArray()]
      else:
        img0.append(image0.GetArray())
      grabResult.Release()
    
    img = img0.copy()
    for i in range(len(self.devices)):
      img[i]=img0[order[i]]
    
    self.cameras.StopGrabbing()
    self.numberOfFramesGrabbed += 1
    if self.numberOfFramesGrabbed > 1:
      print('{} frames grabbed'.format(self.numberOfFramesGrabbed))
    else:
      print('{} frame grabbed'.format(self.numberOfFramesGrabbed))
    return img
```
Once the baslercam object is initialized, grabImageToArray is called over and over again in order to update the live images. All works fine - at first. But after some time, one of the cameras (not always the same one) crashes. There is no error message; the camera just won't send another image and is skipped by the RetrieveResult command, which just gets an image from the next camera instead, and GrabSucceeded returns True. For example, if the above code is run with two cameras present and one of them crashes, grabImageToArray will just return the image from the still working camera twice.
Restarting the program does not solve the problem. Also, if I open the crashed camera with pylon Viewer and try to take pictures, there is no error message, but I don't get any pictures either.
The only way I've found to make a crashed camera work again is cutting off the camera's power supply and reconnecting it.
Now does anyone have an idea what makes the cameras crash and how I could avoid it? Is there perhaps some sort of memory leak in the code?

thiesmoeller, 2019-10-17:
What is your OS and what camera you are using ?

lusaeg, 2019-10-18:
I'm running the code on Ubuntu 18.04.3 and on Windows 7 Professional N. The error occurs on both computers. The camera model is Basler acA1300-30gm.

thiesmoeller, 2019-10-21:
One comment to your code:
You are opening and closing the cameras for every frame you capture.
As this is very inefficient,you should restructure to only open once
 and just retrieving the frames in your method

lusaeg, 2019-10-25:
Thanks! That was actually what the problem was. Turns out that calling StartGrabbing() and StopGrabbing() with every frame is not only inefficient, but also makes the cameras crash. I'm just opening the camera in the beginning now and it seems to be working.

closed 2019-10-25 by lusaeg