basler/pypylon #161: _genicam.RuntimeException: Failed to open 'Basler Camera'

BahadirGLCK, 2019-10-19:
I'm using a Basler GigE camera that is shown on the localhost web page by apache, flask, and OpenCV. Also, I capture an image with pylon and convert the image for OpenCV format. 

I open the localhost on any browser and the page is coming. I can see things expected. When I click the 'Start' button for streaming, I can see that stream. But when ı refresh stream page or click the 'Start' and I get the below error. 

here's the part where I get an error in code. Specifically  `camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)`
```
  class CaptureCamera(BaseCamera):

    @staticmethod
    def frames():
        camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice()
        camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)  
        converter = pylon.ImageFormatConverter()
        converter.OutputPixelFormat = pylon.PixelType_BGR8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

        statepfcapture = False
        counter = 0

        while camera.IsGrabbing():   
            ...
            ...
```
I think that I start a camera with thread when I refresh the page or twice, three times the Camera is not reopening.  How can I fix this problem? 

------------------------------------------------------------------------------------------------------------------------------------
OpenCV version: 4.1.0
Flask Version: 1.1.0
Python Version: 3.6.8
Aboth System:  Ubuntu 18.04 LTS and JetsonTX2 Developer Kit
Camera: Basler acA1300-60gc

Error:

```
Starting camera thread.
Starting camera thread.
Exception in thread Thread-1:
Traceback (most recent call last):
File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
self.run()
File "/usr/lib/python3.6/threading.py", line 864, in run
self._target(*self._args, **self._kwargs)
File "/var/www/tuvis/base_camera.py", line 93, in _thread
for frame in frames_iterator:
File "/var/www/tuvis/camera_basler.py", line 49, in frames
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
File "/usr/local/lib/python3.6/dist-packages/pypylon/pylon.py", line 2919, in StartGrabbing
return _pylon.InstantCamera_StartGrabbing(self, *args)
 _genicam.RuntimeException: Failed to open 'Basler acA1300-60gc#0030532FB8CD#169.254.206.184:3956'. The device is controlled by another application. Err: An attempt was made to access an address location which is currently/momentary not accessible. (0xE1018006)
RuntimeException thrown (file 'GigEDevice.cpp', line 264)
Stopping camera thread due to inactivity.`
```

thiesmoeller, 2019-10-21:
You open the camera multiple times, possibly from multiple threads.

In the frames method the camera has to be Closed after all frames are captured

BahadirGLCK, 2019-10-21:
I tried but the system has already worked from just flask and close all frames. When I run the system with apache I get this error.

thiesmoeller, 2019-10-21:
The software stack you are using is not just pypylon, as you write....

As I don't know how your software is structured it is hard to say why the camera is opened twice ...

In your printout the camera thread message is coming twice ...

closed 2020-12-02 by thiesmoeller