basler/pypylon #163: Access cameras through ubuntu 18.04 in windows.

SpangeJ, 2019-10-24:
I am running a Linux distribution in Windows 10 which I installed through App Store.
The setup is described here: 
[Windows Subsystem for Linux Installation Guide for Windows 10](https://docs.microsoft.com/en-us/windows/wsl/install-win10)
I have two cameras connected and I can see them in pylon viewer and python on windows,
`from pypylon import pylon`
`devices = pylon.TlFactory.GetInstance().EnumerateDevices()`
`len(devices)`
prints: `2`
If I run the same commands in the ubuntu18.04 terminal it prints `0`
So that the terminal cannot find the cameras.

Any thoughts?

kobotschick, 2020-07-25:
Any solutions for using pylon in WLS ?

benedikth, 2021-05-12:
As far as I know, neither wsl nor wsl2 supports usb camera's. Therefore pylon as well as pypylon cannot access the camera if wsl/ wsl2 is used (you'll find a lot of opened issues concerning this topic).

thiesmoeller, 2025-12-15:
Closing as answered. WSL (Windows Subsystem for Linux) and WSL2 do not support USB camera access. This is a limitation of WSL, not pypylon. USB cameras cannot be accessed from within WSL because WSL doesn't have direct access to USB hardware. To use pypylon with USB cameras, you need to run on native Linux, not WSL.