basler/pypylon #164: Using pypylon on raspberry pi 4 or other arm64?

Dev-Solitech, 2019-10-29:


thiesmoeller, 2019-11-15:
pypylon works perfectly on raspberry pi 4 and other arm64 platforms

For wheels check in https://github.com/basler/pypylon/releases
e.g.
https://github.com/basler/pypylon/releases/download/1.5.1/pypylon-1.5.1-cp37-cp37m-linux_aarch64.whl

VervenneThomas, 2020-01-21:
@thiesmoeller Hi, I tried using the wheel you suggested on a raspberry pi4. Unfortunately the wheel is not supported.

thiesmoeller, 2020-01-21:
It was a 'e.g.' wheel.
Check the python version on your rpi4.
The wheel is named accordingly. E.g. 
cp37 stands for python 3.7

james0807, 2020-03-04:
@thiesmoeller Hi, I'm having the same issue, my python version is 3.7 on the Pi 4, but I'm getting the "not supported on this platform error". I think the Pi is a 32 bit system, could that be the problem?

james0807, 2020-03-04:
@VervenneThomas Hi, did you get anywhere with this? Did you get any of the wheels to install properly on the Pi 4?

thiesmoeller, 2020-03-04:
As we only build for arm64 ... Just compile the module

james0807, 2020-03-04:
Okay so there are no pre-built wheels for Raspberry Pi 4? Doesn't bother me just making sure I've understood and that this is clear in case anyone comes looking for something similar.

Thanks :)

thiesmoeller, 2020-03-04:
No. You asked for arm32 binaries...
Pi4 with arm64 is available ....

james0807, 2020-03-05:
Not officially though? Raspbian is 32 bit unless i'm completely lost!

Dev-Solitech, 2020-03-05:
[image: image.png]
Raspberry pi 4 using this wheels.
cp3.... is your python version.
Hope this helps.

thiesmoeller, 2020-03-05:
As I use all boards on aarch64 ... Slighty misinformed. Check the armv7 builds. These are your arm32 bit builds

james0807, 2020-03-05:
Excellent thank you. How are you running on 64 bit?

thiesmoeller, 2020-03-09:
There are multiple distributions that provide 64 bit. Raspian main is still
on 32bit for compatibility reasons.

closed 2020-12-01 by thiesmoeller