basler/pypylon #166: Upload linux whl to pypi

krrishnarraj, 2019-10-31:
https://pypi.org/project/pypylon/#files has only windows whls uploaded.
It would be better to also have linux whls as in github release assets

basler-oss, 2019-11-14:
We didn't upload linux wheels to pypi, because at the time the build system was setup pypi didn't yet allow aarch64 wheels. We choose by "either all or none" an therefore none. With https://www.python.org/dev/peps/pep-0599/ that changed and it is now possible to upload aarch64 binaries to pypi. As soon as someone finds time to implement that, we will upload them to pypi.

JohannLange, 2020-05-18:
Any updates on this issue?

stefanklug, 2020-08-28:
Unfortunately the newest pylon release has binary dependencies that are newer than the ones specified in manylinux214. So currently we are not able to upload pypylon wheels to pypi. We are tracking this internally, but I don't expect a change here soon.

marv, 2021-02-24:
manylinux_2_24 / PEP600 brought the anticipated change and pypylon wheels for all platforms are now available from https://pypi.org/project/pypylon/#files

closed 2021-03-01 by krrishnarraj