basler/pypylon #167: pypylon: Error while retreiving image | AttributeError: 'PylonImage' object has no attribute 'GetImageBuffer'

sidify, 2019-11-06:
Hi, I am using pypylon with BASLER acA2500-20gc to get an image using opencv according to the sample program provided. However getting this error in line 
img = image.GetArray()  

 **AttributeError: 'PylonImage' object has no attribute 'GetImageBuffer'**.

Can you please tell me what is wrong here? 

Regards

'''
A simple Program for grabing video from basler camera and converting it to opencv img.
Tested on Basler acA1300-200uc (USB3, linux 64bit , python 3.5)
'''
from pypylon import pylon
import cv2

# conecting to the first available camera
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

# Grabing Continusely (video) with minimal delay
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        **img = image.GetArray()**
        cv2.namedWindow('title', cv2.WINDOW_NORMAL)
        cv2.imshow('title', img)
        k = cv2.waitKey(1)
        if k == 27:
            break
    grabResult.Release()
    
# Releasing the resource    
camera.StopGrabbing()

cv2.destroyAllWindows()

thiesmoeller, 2019-11-06:
```
diff --git a/src/pylon/PylonImage.i b/src/pylon/PylonImage.i
index 24d108e..2dceea8 100644
--- a/src/pylon/PylonImage.i
+++ b/src/pylon/PylonImage.i
@@ -58,7 +58,7 @@
             shape, dtype, format = self.GetImageFormat(new_pt)
         else:
             shape, dtype, format = self.GetImageFormat(pt)
-            buf = self.GetImageBuffer()
+            buf = self.GetBuffer()
 
         # Now we will copy the data into an array:
         return _pylon_numpy.ndarray(shape, dtype = dtype, buffer=buf)
-- 
```

sidify, 2019-11-11:
Thanks this works with this change.
Are there going to be a fix applied for this in the master?

thiesmoeller, 2020-12-01:
is fixed in release

closed 2020-12-01 by thiesmoeller