basler/pypylon #168: how to take photo from 2 camera with pypylon

ababayi, 2019-11-08:
Hi
I have written a program for taking photos from 2 camera but when it runs , it return a error.
I want to know what the problem of this program :
`
# Import Libraries
from pypylon import pylon
import msvcrt
import numpy as np
import cv2
import time

# Configurations
Camera_1_serial = "22326007"
Camera_2_serial = "22476015"
uDelayTime = 1000000

# Camera Info Variables
Camera_1_info = None
Camera_2_info = None

# Getting Camera Info : In this Step the list of cameras will extract and extract device info
for i in pylon.TlFactory.GetInstance().EnumerateDevices():
    # Camera 1 finder
    if i.GetSerialNumber() == Camera_1_serial:
        Camera_1_info = i
        break

    # Camera 2 finder
    if i.GetSerialNumber() == Camera_2_serial:
        Camera_2_info = i
        break
else:
    print('Error in finding !')

# Opening cameras if the have find
if Camera_1_info is not None and Camera_2_info is not None:
    camera1 = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateDevice(Camera_1_info))
    camera2 = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateDevice(Camera_2_info))
else:
    print("Opening cameras failed !")

# Grabing from cameras Continusely (video) with minimal delay
camera1.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
camera2.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()  # Setup a Conventor

# Setup the conventor parameters Converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

# Main program loop (in this loop frames will be grabbed in each cycle)
i = 0  # iterator
while camera1.IsGrabbing() and camera2.IsGrabbing():

    # Loop Breaker
    if msvcrt.kbhit():
        if ord(msvcrt.getch()) == 27: # If you press Ecs btn the loop will break
            break
    grabResult1 = camera1.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)  # The camera 1 data
    grabResult2 = camera1.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)  # The camera 2 data

    if grabResult1.GrabSucceeded() and grabResult2.GrabSucceeded():
        # Access the image data of camera 1
        image1 = converter.Convert(grabResult1)
        img1 = image1.GetArray()

        # Delay shotting
        time.sleep(uDelayTime / 1000000.0)

        # Access the image data of camera 2
        image2 = converter.Convert(grabResult2)
        img2 = image2.GetArray()

        # Saving data
        cv2.imwrite('grabs/main/cam1/cam1-{}.bmp'.format(i), img1)  # saving the image grabbed from camera 1
        cv2.imwrite("grabs/main/cam2/cam2-{}.bmp".format(i), img2)  # saving the image grabbed from camera 2

    grabResult1.Release()
    grabResult2.Release()
    i = i+1

# Releasing the resource
camera1.StopGrabbing()
camera2.StopGrabbing()
`
thanks :)

thiesmoeller, 2019-11-08:
"it return a error."
what is the error message, that you see?

binglebai, 2020-01-09:
Have you solved the problem
I had a similar problem with connecting the two cameras

ababayi, 2020-01-10:
@stefanklug do you know how we can solve the problem ?

thiesmoeller, 2020-01-10:
@ababayi ... What is the error message that you get?