basler/pypylon #169: Image acquisition from Basler acA2040-55uc has been failing intermittently

prasadpant, 2019-11-08:
Camera is mounted on vehicle and we are trying to capture images while in motion and when grabbing images it is throwing following error intermittently "**Image acquisition on Basler acA2040-55uc failed! Error: 0xe200000f "Device has been removed from the PC**".  Please  see the attached image. Error can be seen on Pylon Viewer itself or from the Python program used for grabbing images.
[image]

Basler Model : acA2040-55uc USB 3.0 Camera 
Operating System: Ubuntu: 16.04
Cable USB 3.0, Micro B screw lock/A, 5m Data Cable.

sith228, 2019-11-20:
How do you solve this one?

thiesmoeller, 2025-12-15:
Closing as answered. The "Device has been removed from the PC" error (0xe200000f) is a hardware/connection issue, not a pypylon bug. This typically occurs when:
- USB cable connection is loose or intermittent (especially with 5m cables)
- USB port power is insufficient
- Cable quality is poor or damaged
- Physical connection is disrupted during operation
For vehicle-mounted cameras, ensure secure cable connections, use high-quality USB 3.0 cables, and check USB port power delivery. This is a hardware reliability issue that needs to be addressed at the physical connection level.

closed 2025-12-15 by thiesmoeller