basler/pypylon #170: Modul not found after installation

yuzouDE, 2019-11-08:
After all installation I try to test grab.py, but get this error msg:
`Traceback (most recent call last):
  File "grab.py", line 14, in <module>
    from pypylon import pylon
ModuleNotFoundError: No module named 'pypylon'`

But I did installed pypylon. So I verify it by `pip3 freeze`, I saw:
`pypylon==1.4.0`

I really have no clue why python3 cannot find this modul, pls help!
Thx in advance!

seysn, 2019-11-08:
Are you sure you're running `grab.py` with python3 ?

yuzouDE, 2019-11-11:
> Are you sure you're running `grab.py` with python3 and not python2 ?

yep. I am using 
`python3 grab.py`

Curmen, 2019-12-03:
What is your python version ?

SMA2016a, 2020-07-14:
most properly the pypylon has been installed in python2  environment.
a quick test would be trying following command.
python -c "import pypylon.pylon as py; print(py.version)"