basler/pypylon #171: Quick change of Exposure time for Basler acA1300-30gm camera (GigE)

lusaeg, 2019-11-08:
Hi there,
I'm programming a GUI, where live Images from several cameras (Basler acA1300-30gm) are displayed simultanousely. Upon a user command, the current live pictures are saved to files. Due to the specific application of the program, it is important to have a relatively long exposure time for the pictures that are saved. For the live images, however, the exposure time should be a lot shorter, to allow for a higher frame rate.
My current implementation basically works, but has a major drawback. I'm changing the exposure time according to the sample code parametrizecameraloadandsaveconfig.py, where the node map is modified and then loaded to the cameras. This step takes about one second for each camera. Each time a set of pictures is saved, this is done twice. For that reason, saving a set of pictures takes around 8 to 10 seconds, which is unacceptable.
So I'm looking for a more efficient way to change from one exposure time to another. I found issue #67 and tried to solve the problem based on @thiesmoeller's answer, but I couldn't make it work. When I tried running a script that only consisted of @thiesmoeller's code from issue #67 (except for the line `info.SetDeviceClass("BaslerUsb")`, which I replaced by `info.SetDeviceClass("BaslerGigE")`) I received the following error:
Traceback (most recent call last):
  File "...\expTime.py", line 27, in <module>
    cam.ChunkSelector = "ExposureTime"
  File "...\Python37\lib\site-packages\pypylon\pylon.py", line 4045, in __setattr__
    self.GetNodeMap().GetNode(attribute).SetValue(val)
  File "...\Python37\lib\site-packages\pypylon\genicam.py", line 2443, in SetValue
    return _genicam.IEnumeration_SetValue(self, entry)
_genicam.InvalidArgumentException: Feature 'ChunkSelector' : cannot convert value 'ExposureTime', the value is invalid. : InvalidArgumentException thrown in node 'ChunkSelector' while calling 'ChunkSelector.FromString()' (file 'enumeration.cpp', line 134)

Does anyone know what the problem might be?

PolymaTh-EE98, 2019-11-08:
try ExposureTimeAbs

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
camera.ExposureTimeAbs = 24587

thiesmoeller, 2019-11-08:
@lusaeg  the sequencer mode example is really different for a gige camera and possibly more than what you want

If you only want to change the exposuretime sometime during your acquisition you can simply use what @PolymaTh-EE98  wrote anytime during acquisition:

`camera.ExposureTimeAbs = 42`

The only issue with this is, that you have to know when you get the image with this new exposure time.
To solve this your camera has the chunk feature. 
see also in the basler documentation:
https://docs.baslerweb.com/data-chunks.html#exposure-time-chunk

```
# Enable data chunks
camera.ChunkModeActive.SetValue(True);
# Select and enable Exposure Time chunk
camera.ChunkSelector.SetValue("ExposureTime");
camera.ChunkEnable.SetValue(True);
```

The camera will attach the exposuretime that was used for this shot to the GrabResult.
For an example how to access chunks see:
https://github.com/basler/pypylon/blob/master/tests/pylon_tests/gigE/chunkimagetest.py

If you detect, that the image has the exposuretime you set, you can save it ...

lusaeg, 2019-11-11:
Thanks to both of you, that's already very helpful.

I have problems implementing the chunk thing suggested by @thiesmoeller, though. Here's how I'm initializing my cameras:
```
tlFactory = pylon.TlFactory.GetInstance()
devices = tlFactory.EnumerateDevices()
cameras = pylon.InstantCameraArray(len(devices))
for i, cam in enumerate(cameras):
        cam.Attach(tlFactory.CreateDevice(devices[i]))
        # Enable data chunks
        cam.ChunkModeActive.SetValue(True);
        # Select and enable Exposure Time chunk
        cam.ChunkSelector.SetValue("ExposureTime")
        cam.ChunkEnable.SetValue(True)
```
When this code is run, the line `cam.ChunkModeActive.SetValue(True);` throws the following exception:

Node is not writable. : AccessException thrown in node 'ChunkModeActive' while calling 'ChunkModeActive.SetValue()' (file 'booleant.h', line 61)

If I evaluate `genicam.IsWritable(cam.ChunkModeActive)` right before that line, it returns `False`. Now according to the error message implemented in the sample file grabchunkimage.py for that case, this means that "The camera doesn't support chunk features". However, when I run grabchunkimage.py, which is only for a single camera, it works, which means that my cameras actually do support chunk features.
Is it possible to use chunk mode for an InstantCameraArray?

Once I've called  `camera.ExposureTimeAbs = 42`, I can't just assume that the following images will be grabbed with exposure time 42, can I?

thiesmoeller, 2019-11-11:
>Once I've called camera.ExposureTimeAbs = 42,
> I can't just assume that the following images will be grabbed with exposure time 42, can I?

No. Because a frame could be in exposure with the old exposuretime while you write the new exposuretime to the camera.
So the "next" frame you receive _after_ setting the exposure time could still have the old exposuretime.

In your code sample:
you have to _open_ the devices to be able to write to them.
So change your code to:

```
tlFactory = pylon.TlFactory.GetInstance()

devices = tlFactory.EnumerateDevices()
cameras = pylon.InstantCameraArray(len(devices))
for i, cam in enumerate(cameras):
        cam.Attach(tlFactory.CreateDevice(devices[i]))
        # Open the device for access to features
        cam.Open()
        # Enable data chunks
        cam.ChunkModeActive.SetValue(True);
        # Select and enable Exposure Time chunk
        cam.ChunkSelector.SetValue("ExposureTime")
        cam.ChunkEnable.SetValue(True)
```

lusaeg, 2019-11-11:
Now the line `cam.ChunkSelector.SetValue("ExposureTime")` throws the exception
Feature 'ChunkSelector' : cannot convert value 'ExposureTime', the value is invalid.

lusaeg, 2019-11-14:
Ok, I'm really stuck here. I tried running the sample code grabchunkimage.py, where I replaced line 38 `info.SetDeviceClass("BaslerUsb")` with `info.SetDeviceClass("BaslerGigE")`. This works fine, but when I add the lines
```
camera.ChunkSelector = "ExposureTime"
camera.ChunkEnable = True
```
right after the same was done with `"Timestamp"`, `"Framecounter"` and `"PayloadCRC16"`, The line `camera.ChunkSelector = "ExposureTime"` throws the following exception:
`
An exception occurred. Feature 'ChunkSelector' : cannot convert value 'ExposureTime', the value is invalid. : InvalidArgumentException thrown in node 'ChunkSelector' while calling 'ChunkSelector.FromString()' (file 'enumeration.cpp', line 134)
`
just as in my own code. 
@thiesmoeller can you see what the error might be?

thiesmoeller, 2019-11-14:
Please check with your support contact for a firmware update to your camera.

On a current acA1300-30gm the sample below works ....

```
import pypylon.pylon as py
import time
import traceback

# open device
tlFactory = py.TlFactory.GetInstance()

di = py.DeviceInfo()
di.SetModelName("acA1300-30gm")

cam = py.InstantCamera(tlFactory.CreateDevice(di))
cam.Open()

print(cam.DeviceModelName.GetValue(), cam.DeviceID.GetValue())

# Enable data chunk
cam.ChunkModeActive = True
# Select and enable Exposure Time chunk
cam.ChunkSelector = "ExposureTime"
cam.ChunkEnable = True

class ChunkImageEventHandler(py.ImageEventHandler):
    def __init__(self, std_exp: float, img_save_exp: float, ts_freq: int):
        """
        init for the chunk handling image event handler
        :param std_exp: standard exposure time to use for the display
        :param img_save_exp: special exposure time to use for saving
        :param ts_freq: timestamp frequency of the device
        """
        super().__init__()
        self.std_exp = std_exp
        self.img_save_exp = img_save_exp
        self.waiting_for_save_exp = True

        # record image timestamps
        self.last_timestamp = 0
        self.ts_freq = ts_freq

    def OnImageGrabbed(self, camera: py.InstantCamera, grab_result: py.GrabResult) -> bool:
        """
        image event handler
        :param camera: reference to the camera object
        :param grab_result: grab result data structure
        :return: false -> stopgrabbing / true -> keep on grabbing
        """
        try:
            if grab_result.GrabSucceeded():
                chunk_exposure = grab_result.ChunkExposureTime.GetValue()

                # calculate timestamp delta to last frame
                timestamp = grab_result.GetTimeStamp()
                delta_to_last_us = (timestamp - self.last_timestamp) / self.ts_freq * 1e6

                if chunk_exposure == self.img_save_exp:
                    if self.waiting_for_save_exp is True:
                        camera.ExposureTimeAbs = self.std_exp
                        self.waiting_for_save_exp = False
                        print("%10.0f" % delta_to_last_us, "save_image with exposure time", chunk_exposure)
                    else:
                        # skip next frame that has  still been acquired with longer exposure time
                        pass
                else:
                    print("%10.0f" % delta_to_last_us, "display image with exposure time", chunk_exposure)
                    self.waiting_for_save_exp = True

                self.last_timestamp = timestamp
                return True
            # stop acquisition in case of error
            print("Error", grab_result.GetErrorDescription())
        except:
            # trace error in python callback handler
            traceback.print_exc()
        return False

# setup the background capture handler
handler = ChunkImageEventHandler(42, 100000, cam.GevTimestampTickFrequency.GetValue())
cam.RegisterImageEventHandler(handler, py.RegistrationMode_Append, py.Cleanup_Delete)

# set the default exposure time
cam.ExposureTimeAbs = 42

# start grabbing continuously
cam.StartGrabbing(py.GrabStrategy_OneByOne, py.GrabLoop_ProvidedByInstantCamera)

while cam.IsGrabbing():
    time.sleep(0.5)
    # write different image save exposure time every 500ms
    time.sleep(0.5)
    # it will be reset to the normal in the capture thread
    cam.ExposureTimeAbs = 100000

cam.StopGrabbing()

```

lusaeg, 2019-11-26:
@thiesmoeller thanks for the tip with the firmware and the sample code. I now managed to update the firmware on one of the cameras from version v3.4-5 to v3.7-16, which apparently is the current version. However, I still have the same problem. The exception I described in my last comment still occurs both in my code and in your sample code. I ran the code on Windows and Ubuntu, both with the same result.
What else could be the problem?

thiesmoeller, 2020-12-01:
have you found a solution already?

thiesmoeller, 2022-03-23:
Closed due to no activity

closed 2022-03-23 by thiesmoeller