basler/pypylon #172: How to make the program like ParameterizeCameraSample

HajimeMiyata, 2019-11-13:
I want to make py sample that can access camera's all parameters like cpp sample program ParametrizeCamera_GenericParameterAccess.cpp.

We want to change the setting of ExposureAuto and WhiteBalanceAuto via python.

[ParametrizeCamera_GenericParameterAccess.txt](https://github.com/basler/pypylon/files/3840682/ParametrizeCamera_GenericParameterAccess.txt)

thiesmoeller, 2019-11-13:
Please take a look into the samples or the readme.

You can access all parameters directly by its name.
e.g.
`camera.ExposureAuto = "Off"`
or
`camera.Width = 1920`

The names are completely identical to C++

HajimeMiyata, 2019-11-14:
[grab_ExposureAuto_WhiteBalanceAuto.txt](https://github.com/basler/pypylon/files/3844686/grab_ExposureAuto_WhiteBalanceAuto.txt)

Thank you for your answer!.
Now, I can make python program to set ExposureAuto and WhiteBalanceAuto at Once.

closed 2020-12-01 by thiesmoeller