basler/pypylon #173: Multicast with Gige ethernet samples?

mbufi, 2019-11-20:
Is there any samples on this?

mbufi, 2019-11-28:
Anyone attempt this?

IceTDrinker, 2019-11-29:
I'm also interested if you have leads on how to do this.

IceTDrinker, 2019-12-01:
see 2.5 here : https://www.baslerweb.com/fp-1566549625/media/downloads/documents/users_manuals/AW00148802000_pylon_SDK_Samples_Manual.pdf

The first user needs to open the camera in control mode, then other users can open the camera in multicast mode, careful with bandwidth

mbufi, 2019-12-02:
Good find @IceTDrinker ... So we can confirm it works with C++. 

Any luck finding examples with Python?

davidi-theother, 2019-12-04:
After a little bit of testing that is working for me (use ConfigurationEventPrinter() from samples )

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.MonitorModeActive = True    
    
camera.RegisterConfiguration(ConfigurationEventPrinter(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)

camera.Open()
camera.StartGrabbing()

mbufi, 2019-12-04:
Hi @davidi-theother  I am not really sure what you mean by your response.

What about the 3 scripts in samples? 
`gige_action_command.py`
`gige_announce_renounce.py`
`gige_ip_config.py` 

Would those be used at all with the samples @IceTDrinker provided?

IceTDrinker, 2019-12-04:
I believe there is a nearly 1:1 python code for this, I was not the one to try it. But I believe the snippet from @davidi-theother is basically what we use. Only problem with this solution is apparently bandwidth if camera streams raw images. ConfigurationEventPrinter can be substituted for None safely I think ? Haven’t played much with configurations so feedback is welcome.

mbufi, 2019-12-05:
@davidi-theother So the code you wrote would be for application B correct? Application A (controller) would do what specifically? 

Also, I believe this is just for multicasting? How about if we wanted application A (controller) to only have the camera multicast a single frame to the other applications (clients) at a time?

PartyTrix, 2020-11-10:
> After a little bit of testing that is working for me (use ConfigurationEventPrinter() from samples )
>
> camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice()) [...]

A long search ended here, thanks!! 😄 

For future reference: this got it working for me:

```
from pypylon import pylon
from samples.configurationeventprinter import ConfigurationEventPrinter
import cv2

#load converter
converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

#initate camera in MonitorMode 
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.MonitorModeActive = True
camera.RegisterConfiguration(ConfigurationEventPrinter(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)

#open camera stream
camera.Open()
camera.StartGrabbing()

#while grabbing, display the image using openCV
while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        cv2.namedWindow('title', cv2.WINDOW_NORMAL)
        cv2.imshow('title', img)
        k = cv2.waitKey(1)
        if k == 27:
            break
    grabResult.Release()

camera.Close()
```

closed 2025-12-15 by thiesmoeller