basler/pypylon #174: Module not found: No module named 'pypylon

ljc1313, 2019-11-21:


thiesmoeller, 2019-11-24:
What is your IDE ?
How did you install pypylon ?

ljc1313, 2019-11-25:
I am using IDLE (Python Shell 3.8)

thiesmoeller, 2019-11-25:
How did you install pypylon ?

thiesmoeller, 2019-11-25:
> I am using win 64, and Python 3.6.8

.... pypylon-1.5.1-cp33pcp33m-win32.whl
is pypylon for 
cp33 [ cpython version 3.3 ]

and the win32 .... says its for win 32

The installation is described in the readme on : https://github.com/basler/pypylon

BUT:
at the moment we have no binary built for python 3.8
So you have two options:

a) compile yourself from source ....
b) uninstall the python3.8 install and install a python 3.7 
   this is still fully maintained ...
 https://www.python.org/downloads/release/python-375/)
 and then you can use this wheel 
https://github.com/basler/pypylon/releases/download/1.5.1/pypylon-1.5.1-cp37-cp37m-win_amd64.whl

ljc1313, 2019-11-25:
Ok 

I will install win 64 version, pypylon-1.5.1-cp36-cp36m-win_amd64.whl

I will uninstall python 3.8 and go to 3.7

thiesmoeller, 2019-11-25:
no ;-)

pypylon-1.5.1-cp36-cp36m-win_amd64.whl
This is for python 3.6
you need 
pypylon-1.5.1-cp37-cp37m-win_amd64.whl

ljc1313, 2019-11-25:
Ahh, I see

ran pip3 install pypylon-1.5.1-cp37-cp37m-win_amd64.whl

and pypylon installed successfully

thanks thiesmoller,
 
I will see if grab.py will run now

ljc1313, 2019-11-25:
I had to make a couple additional installs to get grab.py to run normally
1) had to run (in cmd)
python.exe -m pip install numpy 
and
2)had to run (in cmd)
python -m pip install --upgrade pip 
as I had an older version of pip causing issues as well

Thanks thiesmoller

closed 2019-11-25 by ljc1313