basler/pypylon #175: Hardware trigger - grabbing image is not successful every single time

cute-tiger, 2019-11-22:
#### I am setting up a Basler GigE camera model acA1920-40gm for hardware trigger mode using pypylon. The camera is powered by 12v voltage and triggered by a photoeye sensor. 

#### Issue: when the sensor is triggered at a high frequency, such as 4 times per seconds, the camera can grab images successfully, but when the camera is triggered slowly, it fails to grab images. 

# Here are python scripts:

from pypylon import pylon

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
#### To see which device info is available 
print("DeviceClass: ", camera.GetDeviceInfo().GetDeviceClass())
print("DeviceFactory: ", camera.GetDeviceInfo().GetDeviceFactory())
print("ModelName: ", camera.GetDeviceInfo().GetModelName())

#### demonstrate some feature access
new_width = camera.Width.GetValue() - camera.Width.GetInc()
if new_width >= camera.Width.GetMin():
    camera.Width.SetValue(new_width)

numberOfImagesToGrab = 5
camera.StartGrabbingMax(numberOfImagesToGrab)

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data.
        print("SizeX: ", grabResult.Width)
        print("SizeY: ", grabResult.Height)
        img = grabResult.Array
        print("Gray value of 100x100th pixel: ", img[100, 100])

    grabResult.Release()
camera.Close()
print("initial test completed")

#### set up hardware trigger
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

camera.Open()

camera.AcquisitionStatusSelector = "FrameTriggerWait"
camera.TriggerSource.SetValue("Line1")
camera.TriggerMode.SetValue("On")
camera.TriggerActivation.SetValue('FallingEdge') 
camera.TriggerDelayAbs.SetValue(300)
camera.AcquisitionFrameCount.SetValue(1)
camera.PixelFormat.SetValue('Mono8')
camera.TriggerSelector.SetValue('FrameStart')
camera.ExposureTimeRaw.SetValue(800)

camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

count = 0
while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(20000, pylon.TimeoutHandling_ThrowException)
    count += 1
    print("count = ", count)
    if grabResult.GrabSucceeded():
        # Access the image data.
        print("SizeX: ", grabResult.Width)
        print("SizeY: ", grabResult.Height)
        img = grabResult.Array
        print("Gray value of 100x100th pixel: ", img[100, 100])
    else:
        print("grabbing is not successful......")

    grabResult.Release()
camera.Close()

# script ends here!

# Here is output from terminal:

DeviceClass:  BaslerGigE
DeviceFactory:  GigE/BaslerGigE 5.2.0.13457
ModelName:  acA1920-40gm
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  0
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  0
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  0
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  0
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  0
1st test completed
count =  1
grabbing is not successful......
count =  2
grabbing is not successful......
count =  3
grabbing is not successful......
count =  4
grabbing is not successful......
count =  5
grabbing is not successful......
count =  6
grabbing is not successful......
count =  7
grabbing is not successful......
count =  8
grabbing is not successful......
count =  9
grabbing is not successful......
count =  10
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  19
count =  11
grabbing is not successful......
count =  12
grabbing is not successful......
count =  13
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  18
count =  14
grabbing is not successful......
count =  15
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  19
count =  16
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  19
count =  17
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  19
count =  18
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  19
count =  19
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  18
count =  20
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  19
count =  21
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  19
count =  22
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  19
count =  23
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  19
count =  24
SizeX:  1904
SizeY:  1200
Gray value of 100x100th pixel:  18
count =  25
grabbing is not successful......
count =  26
grabbing is not successful......
count =  27
grabbing is not successful......
count =  28
grabbing is not successful......

cute-tiger, 2019-11-23:
# Issue has been resolved by using these codes:
#### Packet Size
camera.GevSCPSPacketSize.SetValue(1500);
#### Inter-Packet Delay
camera.GevSCPD.SetValue(1000);
#### Frame transmission delay
camera.GevSCFTD.SetValue(1000);

closed 2019-11-23 by cute-tiger

MCilento93, 2020-12-27:
> # Issue has been resolved by using these codes:
> #### Packet Size
> camera.GevSCPSPacketSize.SetValue(1500);
> 
> #### Inter-Packet Delay
> camera.GevSCPD.SetValue(1000);
> 
> #### Frame transmission delay
> camera.GevSCFTD.SetValue(1000);

Can you attach the final code with github formatting?