basler/pypylon #177: Cannot install pypylon even when I have swig > 4.0.0

occiAdmin, 2019-11-28:
Cloned the git repo.
performed- pip3 install .
Got the following error.

` Complete output from command python setup.py egg_info:
    Deprecated command line option: -modern. This option is now always on.
    NodeCallback.i:166: Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
    IPort.i:23: Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
    /opt/pylon5/include/GenApi/IBase.h:57: Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
    PortImpl.i:52: Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
    /opt/pylon5/include/GenApi/IBase.h:57: Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
    Deprecated command line option: -modern. This option is now always on.
    /opt/pylon5/include/pylon/DeviceInfo.h:74: Error: Syntax error in input(1).
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-jgyqjc21-build/setup.py", line 691, in <module>
        version
      File "/tmp/pip-jgyqjc21-build/setup.py", line 159, in call_swig
        subprocess.check_call(call_args, cwd=os.path.abspath(sourcedir))
      File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/usr/local/bin/swig', '-python', '-outdir', '/tmp/pip-jgyqjc21-build/pypylon', '-o', '/tmp/pip-jgyqjc21-build/generated/pylon_wrap.cpp', '-c++', '-Wextra', '-Wall', '-threads', '-modern', '-DSWIG_PYTHON_LEGACY_BOOL', '-py3', '-I/opt/pylon5/include', '-I/opt/pylon5/include', 'pylon.i']' returned non-zero exit status 1.
    ExtraCompileArgs: ['-Wno-unknown-pragmas', '-fPIC', '-g0', '-Wall', '-O3', '-Wno-switch', '-I/opt/pylon5/include']
    ExtraLinkArgs: ['-g0', '-Wl,--enable-new-dtags', '-Wl,-rpath,$ORIGIN', '-L/opt/pylon5/lib64', '-Wl,-E', '-lpylonbase', '-lpylonutility', '-lGenApi_gcc_v3_1_Basler_pylon', '-lGCBase_gcc_v3_1_Basler_pylon']
    LibraryDirs: ['/opt/pylon5/lib64']
    Building version: 1.3.1+pylon5.2.0
    call /usr/local/bin/swig -python -outdir /tmp/pip-jgyqjc21-build/pypylon -o /tmp/pip-jgyqjc21-build/generated/genicam_wrap.cpp -c++ -Wextra -Wall -threads -modern -DSWIG_PYTHON_LEGACY_BOOL -py3 -I/opt/pylon5/include genicam.i
    call /usr/local/bin/swig -python -outdir /tmp/pip-jgyqjc21-build/pypylon -o /tmp/pip-jgyqjc21-build/generated/pylon_wrap.cpp -c++ -Wextra -Wall -threads -modern -DSWIG_PYTHON_LEGACY_BOOL -py3 -I/opt/pylon5/include -I/opt/pylon5/include pylon.i
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-jgyqjc21-build/`

thiesmoeller, 2019-11-28:
you are on a really old tag 1.3.1 just use lastest master ... or tag 1.5.1

occiAdmin, 2019-11-28:
I am there because of Ginicam. It kept on throwing error in my multiprocessed python3 program and worked fine on single thread

thiesmoeller, 2019-11-28:
'It kept on throwing error in my multiprocessed' .... exactly this is one of the fixes..

occiAdmin, 2019-11-28:
I have opened a new issue. Kindly have a look at that.

closed 2019-11-28 by occiAdmin