basler/pypylon #178: Unable to retrive image while using the pypylon api in python3 multiprocessing

occiAdmin, 2019-11-28:
I am using basler 1920-40uc. The code works fine while using in a single-threaded process. But due to the nature of the application, I have to make a process that repeatedly calls camera api and put the image data into a queue.

Below is the wrapper class that I have defined. 
`
import sys, os
from pypylon import pylon
import cv2
import logging
import time,datetime

"""
The camera is connected with the optocoupler which triggers the camera to capture a frame.
This even should be treated as the beginning of the time. 

"""

def date():
	return datetime.datetime.today().strftime('%Y-%m-%d')

def simplyfy(): 
	x = datetime.datetime.now()
	return x.strftime("%Y-%m-%d %H:%M:%S")

class CameraInput():
    """
    camera streamer class
    use mode = "cam" for basler cam's input otherwise use "debug" for a custom video
    use tigger = True for trigger based acquisition else it will return continous stream 
    """
    def __init__(self,mode="cam",trigger=True):
        self.debuggerVideoPath = "../storage/Nov-07-2019/16_18_30.mp4"
        self.mode = mode
        self.trigger = trigger
        self.driver()
        

    def driver(self):

        if self.mode == "debug":
            self.feed = cv2.VideoCapture(self.debuggerVideoPath)
        else:
            if  self.trigger:
                self.feed = Basler(flag=0)
            else:
                self.feed = Basler(flag=0)

    def read(self):
        ret,self.frame = self.feed.read()
        if ret:
            return ret,self.frame
        else:
            return None,None

            
    def stop(self):
        self.feed.release()
        self.feed = None

class Basler:
    def __init__ (self, flag=0,exposure=None):
        self.camera=pylon.InstantCamera (pylon.TlFactory.GetInstance ().CreateFirstDevice ())
        self.camera.Open()
        if exposure is None:
            exposure = 10000
        self.camera.ExposureTime.SetValue(int(exposure))

        if flag ==1:
            self.camera.TriggerSelector.SetValue("FrameBurstStart")
            self.camera.TriggerSource.SetValue("Line1")
            self.camera.TriggerMode.SetValue("On")

        # Grabing Continusely (video) with minimal delay
        self.camera.StartGrabbing (pylon.GrabStrategy_LatestImageOnly)
        self.converter=pylon.ImageFormatConverter ()

        # converting to opencv bgr format
        self.converter.OutputPixelFormat=pylon.PixelType_BGR8packed
        self.converter.OutputBitAlignment=pylon.OutputBitAlignment_MsbAligned
        
        
        
        """
        The code below works and it is similar to the code wrapper in 
        "read" function. Calling that function gives error
        """
        
        # self.some =self.camera.RetrieveResult (5000, pylon.TimeoutHandling_ThrowException)
        # self.image=self.converter.Convert (self.some)
        # self.img=self.image.GetArray ()
        # print(self.img)

    def read (self):

        grabResult=self.camera.RetrieveResult (10000, pylon.TimeoutHandling_ThrowException)

        if grabResult.GrabSucceeded():
            # Access the image data
            self.image=self.converter.Convert(grabResult)
            self.img=self.image.GetArray ()
            return True,self.img
        else:
            return  False, None

    def release (self):
        self.camera.StopGrabbing ()
        del self.camera
`

This is the camera process that uses the above code.
`
from multiprocessing import Process
import sys, os
sys.path.insert(0, os.path.abspath('..'))
import time
from imaging.camera import CameraInput

class CamProcess(Process):
    def __init__(self,queue):
        Process.__init__(self)
        self.data_queue = queue
        self.flag = True
        self.id = None
        self.cam = CameraInput(mode="cam", trigger = False)
    
    def run(self):
        if self.id is None:
            self.id = os.getpid()

        while self.flag:
            ret, frame = self.cam.read()
            if ret:
                self.data_queue.put([self.id,frame])

            else:
                self.data_queue.put(None)

    def stop(self):
        self.flag = False
        self.cam.stop()`

upon calling the function I am getting the below error.

`File "/usr/lib/python3.6/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "sortingProcess.py", line 50, in run
    ret, frame = self.cam.read()
  File "/home/xyz/imaging/camera.py", line 52, in read
    ret,self.frame = self.feed.read()
  File "/home/xyz/imaging/camera.py", line 98, in read
    grabResult=self.camera.RetrieveResult (10000, pylon.TimeoutHandling_ThrowException)
  File "/home/xyz/sortingEnv/lib/python3.6/site-packages/pypylon/pylon.py", line 3299, in RetrieveResult
    return _pylon.InstantCamera_RetrieveResult(self, *args)
_genicam.RuntimeException: Error setting waitable timer 0x04487880: Invalid argument : RuntimeException thrown (file 'WaitableTimer.h', line 201)
`

occiAdmin, 2019-11-28:
This error has taken away a good two days in a high-pressure situation. Any help is appreciated.

thiesmoeller, 2019-11-28:
... you should instantiate the InstantCamera in the `run()` method of the `CamProcess`.

Otherwise the multiprocessing code in python will try to move th InstantCamera between the two processes. Which can't work, because this would require 
* to "pickle" the complete state of the InstantCamera 
* transfer it to the other process
* "unpickle" it
* work with it.

As you only transfer the images through your queue this shouldn't be a problem ...

```
import os
from pypylon import pylon
from multiprocessing import Process

class CamProcess(Process):
    def __init__(self, queue):
        Process.__init__(self)
        self.data_queue = queue
        self.flag = True
        self.id = None
        self.cam = None

    def run(self):
        if self.id is None:
            self.id = os.getpid()
        self.cam = Basler(flag=0)
        while self.flag:
            ret, frame = self.cam.read()
            if ret:
                self.data_queue.put([self.id, frame])

            else:
                self.data_queue.put(None)

    def stop(self):
        self.flag = False
        self.cam.stop()

class Basler:
    def __init__(self, flag=0, exposure=None):
        self.camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

        self.camera.Open()

        # Grabing Continously (video) with minimal delay
        self.camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
        self.converter = pylon.ImageFormatConverter()

        # converting to opencv bgr format
        self.converter.OutputPixelFormat = pylon.PixelType_BGR8packed
        self.converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

        """
        The code below works and it is similar to the code wrapper in 
        "read" function. Calling that function gives error
        """

        # self.some =self.camera.RetrieveResult (5000, pylon.TimeoutHandling_ThrowException)
        # self.image=self.converter.Convert (self.some)
        # self.img=self.image.GetArray ()
        # print(self.img)

    def read(self):
        grab_result = self.camera.RetrieveResult(10000, pylon.TimeoutHandling_ThrowException)

        if grab_result.GrabSucceeded():
            # Access the image data
            py_image = self.converter.Convert(grab_result)
            img = py_image.GetArray()
            return True, img
        else:
            return False, None

    def release(self):
        self.camera.StopGrabbing()
        del self.camera

if __name__ == '__main__':
    from multiprocessing import Queue

    im_q = Queue(maxsize=1)
    cam = CamProcess(im_q)
    cam.start()
    while True:
        data = im_q.get()
        print(data)

```

occiAdmin, 2019-11-29:
Thanks a lot. I couldn't comprehend  what was the reason for the error. I did test it and it is working fine.

closed 2019-11-29 by occiAdmin

SarahChiu, 2022-12-02:
Hi @occiAdmin and @thiesmoeller, thanks for sharing the code and explanation! 

I tried to run the code as @thiesmoeller suggested to collect images from acA4024-29uc. The code is run in Ubuntu 20.04. 
However, the process hangs at this line in class Basler and never continues: 
```python
self.camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
```

When I stopped the program by Ctrl+C, it gave me this traceback: 
```bash
File "/usr/lib/python3.9/multiprocessing/popen_fork.py", line 27, in poll
  pid, sts = os.waitpid(self.pid, flag)
```
It seems like the program runs into a deadlock. Any idea why this happens?