basler/pypylon #179: Couldn't import pypylon different directory

Curmen, 2019-11-29:
Hello, I'm trying to setup pypylon but i couldn't figure out. 
Python version is 3.7.4

I download and install pylon software from basler web site.
then i install wheel with `pip install pypylon-1.5.1-cp37-cp37m-win_amd64.whl`

 When my directory on ** C:\ **  i can `import pypylon as py` like shown below
[image]
 
 However when my directory changed **C:\ to C:\pypylon** i couldn't import pypylon 

[image]

 I'm totally confused. Also i already downloaded pypylon folder from github repository and the direction of the folder is **C:\pypylon**
 When i try to run` pip install .` command given this error ;

`ERROR: Complete output from command python setup.py egg_info:
    ERROR: WARNING:root:git not found or invalid tag found.
    WARNING:root:-> Building version from date!
    NodeCallback.i(166) : Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
    IPort.i(23) : Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
    C:\Program Files\Basler\pylon 6\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
    PortImpl.i(52) : Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
    C:\Program Files\Basler\pylon 6\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
    C:\Program Files\Basler\pylon 6\Development\include\pylon\gige\ActionTriggerConfiguration.h(71) : Warning 325: Nested struct not currently supported (ActionParameter ignored)
    ..\genicam\NodeCallback.i(166) : Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
    ..\genicam\IPort.i(23) : Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
    C:\Program Files\Basler\pylon 6\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
    ..\genicam\PortImpl.i(52) : Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
    C:\Program Files\Basler\pylon 6\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
    C:\Program Files\Basler\pylon 6\Development\include\pylon\PylonImageBase.h(38) : Warning 403: Class 'Pylon::CPylonImageBase' might be abstract, no constructors generated,
    C:\Program Files\Basler\pylon 6\Development\include\pylon\ReusableImage.h(50) : Warning 403: Method Pylon::IReusableImage::IsSupportedPixelType(Pylon::EPixelType) const might not be implemented.
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\setup.py", line 900, in <module>
        bs.copy_runtime()
      File "C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\setup.py", line 481, in copy_runtime
        super(BuildSupportWindows, self).copy_runtime()
      File "C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\setup.py", line 211, in copy_runtime
        shutil.copytree(src, dst)
      File "C:\Users\SpechtLab_Mec\Anaconda3\lib\shutil.py", line 315, in copytree
        names = os.listdir(src)
    FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Program Files\\Basler\\pylon 6\\Development\\..\\runtime\\x64\\pylonCXP'
    Building version: 2019.11.29.dev60505
    Remove: .\generated
    Remove: .\pypylon
    call C:\swigwin-3.0.12\swig.exe -python -outdir C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon -o C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\generated\genicam_wrap.cpp -c++ -Wextra -Wall -threads -py3 -DHAVE_PYLON_GUI -D_WIN32 -D_WIN64 -IC:\Program Files\Basler\pylon 6\Development\include genicam.i

    call C:\swigwin-3.0.12\swig.exe -python -outdir C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon -o C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\generated\pylon_wrap.cpp -c++ -Wextra -Wall -threads -py3 -DHAVE_PYLON_GUI -D_WIN32 -D_WIN64 -IC:\Program Files\Basler\pylon 6\Development\include -IC:\Program Files\Basler\pylon 6\Development\include pylon.i

    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\PylonGUI_v6_0.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\PylonUtility_v6_0.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\PylonBase_v6_0.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\GCBase_MD_VC141_v3_1_Basler_pylon.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\GenApi_MD_VC141_v3_1_Basler_pylon.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\log4cpp_MD_VC141_v3_1_Basler_pylon.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\Log_MD_VC141_v3_1_Basler_pylon.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\NodeMapData_MD_VC141_v3_1_Basler_pylon.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\XmlParser_MD_VC141_v3_1_Basler_pylon.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\MathParser_MD_VC141_v3_1_Basler_pylon.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\uxapi_v11.dll => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\
    Copy tree C:\Program Files\Basler\pylon 6\Development\..\runtime\x64\pylonCXP => C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\pypylon\pylonCXP
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in C:\Users\SPECHT~1\AppData\Local\Temp\pip-req-build-rosa2ft3\`

PolymaTh-EE98, 2019-11-29:
For most system pip installs in python 2 and pip3 installs in python 3

Then to run python 3 use python3  not python which is python 2

Curmen, 2019-12-01:
> For most system pip installs in python 2 and pip3 installs in python 3
> 
> try typing pip -V
> 
> you should get something like this below
> 
> ~$ pip -V
> pip 8.1.1 from /usr/local/lib/python2.7/site-packages (python 2.7)
> 
> and pip3 -V
> 
> ~$ pip3 -V
> pip 8.1.1 from /usr/local/lib/python3.5/site-packages (python 3.5)

Thank you PolymaTh-EE98. When i install python on Conda Virtual env. , i have pip3 
My Conda Enviroment's python version is 3.5.6
Which i try to install the version of python 3.7.4 so i guess this may cause the problem.

closed 2019-12-01 by Curmen