basler/pypylon #181: How can i set exposure Time via python

Curmen, 2019-12-03:
Hello,
 I try to understand basler raL2048-48gm these couple of days. So every i plug off and on the power in camera. Exposure time set 14 ms. But i want to configure just once and the exposure time should be same when i power off and on the camera.  
 So in my .py application i already try but it didn't help. 

```
camera.AutoExposureTimeAbsLowerLimit.SetValue(1000)
camera.AutoExposureTimeAbsUpperLimit.SetValue(5000)
camera.AutoTargetValue.SetValue(128);
camera.ExposureAuto.SetValue(ExposureAuto_Continuous)

```

thiesmoeller, 2019-12-03:
Read the section about UserSets/Configuration Sets in the camera manual.
You can change settings. Store them in one of three flash locations in the camera and configure your camera to boot the next time with these settings.

thiesmoeller, 2019-12-04:
If you want to let the camera start with a given set of configuration:
e.g. always start with 20ms exposuretime for the next boot

```
camera.ExposureTimeAbs = 20000

camera.UserSetSelector = "UserSet1"
camera.UserSetSave.Execute()
camera.UserSetDefaultSelector = "UserSet1"
```

Curmen, 2019-12-05:
Thank you so much moeller. Both commands help me

closed 2019-12-05 by Curmen