basler/pypylon #182: Swig executable not found on path

lcflorindo, 2019-12-05:
Hi, I'm trying to install pypylon on a SuSe 12 SP4 server, but having some trouble with Swig path:
```
ERROR: Command errored out with exit status 1:
     command: /home/filholf/anaconda3/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-9bp_pfvk/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-9bp_pfvk/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base pip-egg-info
         cwd: /tmp/pip-req-build-9bp_pfvk/
    Complete output (11 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-9bp_pfvk/setup.py", line 803, in <module>
        bs = BuildSupport.make()
      File "/tmp/pip-req-build-9bp_pfvk/setup.py", line 320, in make
        return BuildSupportLinux()
      File "/tmp/pip-req-build-9bp_pfvk/setup.py", line 607, in __init__
        self.SwigExe = self.find_swig()
      File "/tmp/pip-req-build-9bp_pfvk/setup.py", line 124, in find_swig
        raise RuntimeError("swig executable not found on path!")
    RuntimeError: swig executable not found on path!
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

Which is the correct path and how to set it?

Thanks in advance!

thiesmoeller, 2019-12-05:
http://www.linfo.org/path_env_var.html

You have to be able to call 'swig' from the environment from where you call setup.py.

Check that you use swig4.x ..

Curmen, 2019-12-10:
Make sure you have to call "swig" from your enviroment. Install swig with **pip3**

lcflorindo, 2019-12-10:
I did it by installing swig and gcc by `conda install -c anaconda siwg` and `conda install -c anaconda gcc` directly on my env.

Thanks for the support!

closed 2019-12-10 by lcflorindo