basler/pypylon #183: OutOfRangeException when trying to change exposure time

Scoooooba, 2019-12-09:
Hey,
a problem with a python script that I created a few months ago has been reported to me. I don’t have physically access to the camera and computer used so I can’t really try to debug on my own, that’s why I’d appreciate your help, maybe you know what to do.

The error occurs when trying to change the exposure time of a certain camera model (currently I don’t know which model but I can find out if it´s important to know). While the script works fine with other camera models, this time the following error is thrown:

`The difference between Value = 109994 and Min = 35 must be dividable without rest by Inc = 35. : OutOfRangeException thrown in node 'ExposureTimeRaw' while calling 'ExposureTimeRaw.SetValue()' (file 'integert.h', line 79)`

Based on the Error I would say the exposure time to set is somehow invalid. But I don´t understand why? Sure, the error says the exposure time value must be dividable without rest by some certain value, but why is it 35 and where does this come from? And why didn’t this error occur with other camera models?

Again, I’d appreciate your help.

Best regards
Scoooooba

thiesmoeller, 2019-12-11:
Machine vision cameras have to be precise.
The model that you have has an increment on the exposuretime of 35us. This is why you see this error.

Scoooooba, 2019-12-11:
Ah thanks. Is there a function to read out the increment of a camera, so i can check it and adjust the exposure time to be set based on it?

MattsProjects, 2019-12-11:
Yup, in the same way there is GetMin() and GetMax(), there should also be a GetInc().

Scoooooba, 2019-12-11:
Perfect, thank you guys. I will try it out as soon as i have access.

closed 2020-12-01 by thiesmoeller