basler/pypylon #184: camera.GetSfncVersion() is not wrapped?

Cody-G, 2019-12-11:
I get a SWIG pointer to VersionInfo:

```<Swig Object of type 'VersionInfo *' at 0x7f9914d00960>```

Does this still need to be wrapped, and what would be the desired python type to return?  A version string?  Thanks!

thiesmoeller, 2019-12-11:
bug in current version
should return this .... https://github.com/basler/pypylon/blob/master/src/genicam/Version.i

pierreyvessanterre, 2021-04-01:
Is this issue still opened? I can't seem to find how to do the PyPylon moral equivalent of C++ of SDK's samples :
            if (camera.GetSfncVersion() < Sfnc_2_0_0)	
To decide to use Gain or GainRaw

yanboyang713, 2021-06-25:
I found function:
```python
camera.GetSfncVersion
```
Does work on  device  a2A1920-51gcPRO

closed 2025-12-15 by thiesmoeller