basler/pypylon #186: Error: 'Device is exclusively opened by another client.'

mattcarp12, 2019-12-13:
`_genicam.RuntimeException: Failed to open device '2676:ba02:2:2:25' for XML file download. Error: 'Device is exclusively opened by another client.' : RuntimeException thrown (file 'PylonUsbTL.cpp', line 474)`

I'm trying to make a class to wrap the `InstantCamera` object, but the problem is if I try creating multiple instances of this object the above exception gets thrown. How can I get around this by just getting a reference to the current instance of the device?

This will be part of a django app. I can make it so this class is instantiated only once in a single process, but if there are multiple django processes it will cause the above error.

Any tips?

Here is my code:

```python
class Camera:
    def __init__(self):
        self.camera = pylon.InstantCamera(
            pylon.TlFactory.GetInstance().CreateFirstDevice()
        )
        self.camera.Open()
        self.camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
```

nmarc, 2019-12-13:
The Problem may be "pylon.TlFactory.GetInstance().CreateFirstDevice()" in
your Class init.

Every instance calls the first device.

mattcarp12, 2019-12-13:
There will only ever be one device attached.

nmarc, 2019-12-15:
If you create multiple instances of this Class, every Class try to Connect
to the First (and only) Camera. The First instance can so this, every
following instance is Blocked. You can only do a single Connection to one
Camera.

thiesmoeller, 2025-12-15:
Closing as answered. USB cameras can only be opened by one client/process at a time. If you create multiple instances of your Camera class, each tries to open the same device, causing the exclusive access error. Solutions:
- Use a singleton pattern to ensure only one camera instance exists
- For Django with multiple processes, use a shared camera service/daemon
- Close the camera properly before opening in another process
This is expected behavior - cameras require exclusive access for proper operation.