basler/pypylon #187: problem in installing pypylon in python 3.8

ali-atghaei, 2019-12-15:
hi. I cant install pypylon neither with pip nor with its binary release. 
in its binary release when i run ( python setup.py install ) it returns error and with (pip install . ) it says "ERROR: Command errored out with exit status 1: python setup.py egg_info Check th
e logs for full command output." 
what should I do for installing pypylon in python 3.8 ? I have force to use python 3.8.
I try each wheel version in pypi.
i have installed visual studio and swig and have added their path in my system. 
thanks for helps.

nmarc, 2019-12-15:
On Linux or Windows?

ali-atghaei, 2019-12-15:
on Windows

nmarc, 2019-12-15:
Can you pleased try winpython
Version "WinPython64-*3.8*.0.0cod"

And Install via pip
( Set Proxy If needed)
C:\python -m pip install pypylon

ali-atghaei, 2019-12-15:
I tried it. says : 
ERROR: Could not find a version that satisfies the requirement pypylon (from ver
sions: none)
ERROR: No matching distribution found for pypylon

nmarc, 2019-12-15:
Which Pylon Version is installed ?

ali-atghaei, 2019-12-15:
pylon V5

amalsafa, 2020-03-11:
@ali-atghaei  I tried to install pypylon on Raspberry but i faced the same issue as you, please did you find a solution for that?

thiesmoeller, 2021-02-23:
Beginning with release 1.7.1 we have binary releases on all platforms, that you can install via pip.

update your pip version before by calling `python -m pip install --upgrade pip`

closed 2021-02-23 by thiesmoeller