basler/pypylon #189: Problems about basler resolution

pch9520, 2019-12-25:
I have a GIGE basler camera.I use pypylon to grab the video stream,but it's frame rate only have 5-7 fps.I find the problem is the image resolution is too high(1920*1200),I want to adjust the resolution to (640*400) by using pylon viewer software.But this software only allow me to crop the image,not to resize the image.
How should I do to resize the image grabbing from GIGE basler camera by using pypylon library?

pch9520, 2020-01-03:
I find my basler's type is aca1920-40gc,it can not resize picture before  spreading into my computer!

nmarc, 2020-01-03:
Try to Set Up with Basler Camara Manager. Save the settings as userSet1 and
also activate "load userSet1 on Camerastart"
If the Problem persists the Feature is Not activated in the Camer

closed 2020-12-01 by thiesmoeller