basler/pypylon #190: Problem about connecting two basler camera

binglebai, 2020-01-07:
I can connect a basler camera based on Win10 but cannot connect two cameras at the same time

thiesmoeller, 2020-01-07:
What Interface?
Have you used the pylon viewer and the integrated tools (ip/usb-)configurator to debug ?

binglebai, 2020-01-08:
Basler camera is  aca1920-40gc  Gige 
I used the pylon viewer can connect the camera but i dont know how to connenting two camera by
python

lcflorindo, 2020-01-27:
I'm facing the same issue, but to use two cameras in different applications.
I tried this code that I found over internet:
```
ip_address = '10.16.194.1'
factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info = ptl.CreateDeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', ip_address)
camera_device = factory.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)
camera.Open()
camera.PixelFormat.SetValue("BayerRG8")
```
But if I connect both cameras at the same IP range, the last connected overlaps the first one, even if i change the `ip_address` to differentiate it.

Now I'm not sure if this code sets the ip on every connection or if "guides" the code to find the camera (like the F12 function on Pylon viewer).

binglebai, 2020-03-23:
i connect two cameras at the same time , i dont know how to solve the problem
`_genicam.RuntimeException: Failed to open 'Basler acA1920-40gc#0030531DC4E7#192.168.1.101:3956'. The device is controlled by another application. Err: An attempt was made to access an address location which is currently/momentary not accessible. (0xE1018006)
 : RuntimeException thrown (file 'gigedevice.cpp', line 264)`

bwllc, 2020-06-23:
@binglebai : I switch between Pylon Viewer, and developing my own camera application.  I get the error you see when I forget to disconnect a camera in Pylon Viewer before running my own application.  The Basler camera driver can only communicate with one client at a time.

SMA2016a, 2024-07-31:
set the ip address of the camera properly. Use the pylon ip configuration tool to setup the camera.

closed 2025-12-15 by thiesmoeller