basler/pypylon #191: Camera Timeout with external trigger

PolymaTh-EE98, 2020-01-08:
I am externally triggering the camera. When the camera is not being triggered the RetrieveResults will timeout.  I can set this to infinity but then is there a seperate timeout or connectivity check if the camera is actually not functioning or disconnected?

thiesmoeller, 2020-01-14:
For USB the plug&play subsystem of your os will notify you.

For gige the camera and your app exchange a heartbeat signal.

You can register callbacks with the configurationevents to get notified  -- check the samples dir for a corresponding sample.

julienGautier77, 2020-04-12:
hi
you can try to use ExecuteSoftwareTrigger() and see if there some data

PolymaTh-EE98, 2020-04-12:
I’m still understanding how I do this. The retrieve results in python is blocking. How do I check the camera?

thiesmoeller, 2020-04-12:
To detect if the camera disconnects  you can register on Device removal callback.

To prevent a blocking foreground thread due to retrieve result waiting, just use the background grab handling. See [grab thread sample](https://github.com/basler/pypylon/blob/3889a2be5710ea06f86821b71895ec1826c1c71d/samples/grabusinggrabloopthread.py#L20)

Zegorax, 2021-05-19:
@thiesmoeller Hello, is the override function name called "OnDeviceRemoval"? I see this function documented nowhere.

I'm guessing the correct implementation would be like this:
```
class CustomImageListener(pylon.ImageEventHandler):
    def __init__(self):
        super(CustomImageListener, self).__init__()

    def OnDeviceRemoval(self, camera):
        print("A camera has been removed!")
```

Am I correct ?

Zegorax, 2021-09-16:
@thiesmoeller Can you please confirm or deny my previous comment?

Zegorax, 2021-09-16:
@stefanklug @RoccoMatano May I ask you to please take look?

thiesmoeller, 2021-09-16:
@Zegorax it is part of the configuration event handling ( see https://github.com/basler/pypylon/blob/458a881268d6d632c19115d193e374926e4bd988/samples/grabstrategies.py#L28)

```python
import pypylon.pylon as py
import time

class DeviceRemovalPrinter(py.ConfigurationEventHandler):
    def OnCameraDeviceRemoved(self, camera):
        print("OnCameraDeviceRemoved event for device ", camera.GetDeviceInfo().GetModelName())

tlf = py.TlFactory.GetInstance()
cam = py.InstantCamera(tlf.CreateFirstDevice())

cam.RegisterConfiguration(DeviceRemovalPrinter(), py.RegistrationMode_Append, py.Cleanup_Delete)

cam.Open()

print(" unplug your device")

while True:
    time.sleep(0.1)

cam.Close()

```

closed 2021-09-16 by thiesmoeller