basler/pypylon #192: _genicam.AccessException

GuilhermeCarvalho1144, 2020-01-13:
I'm trying to set the width and height in the pypylon API, but I'm getting a error like this.

> _genicam.AccessException: Node is not writable. : AccessException thrown in node 'Width' while calling 'Width.SetValue()' (file 'integert.h', line 77).

Camera Information:
- Basler GigE acA1300-60gc

thiesmoeller, 2020-01-13:
Either you haven't called camera.Open() before this call.

Or you try to do this after camera.StartGrabbing(), which is forbidden, because all DMA ressources are already locked.

GuilhermeCarvalho1144, 2020-01-13:
@thiesmoeller Thanks for the reply. I was not calling camera.open (). My code now looks like this:

```
from pypylon import pylon
import cv2

tl_factory = pylon.TlFactory.GetInstance()
# conecting to the first available camera
camera = None
for dev_info in tl_factory.EnumerateDevices():
    if dev_info.GetDeviceClass() == 'BaslerGigE':
        print("using %s @ %s" % (dev_info.GetModelName(), dev_info.GetIpAddress()))
        camera = pylon.InstantCamera(tl_factory.CreateDevice(dev_info))
        break
else:
    raise EnvironmentError("no GigE device found")

camera.Open()
camera.Width.SetValue(1200)
camera.Heigth.SetValue(1024)
camera.OffsetX.SetValue(0)
camera.OffsetY.SetValue(0)

# Grabing Continusely (video) with minimal delay
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
i=0

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        cv2.namedWindow('Debug', cv2.WINDOW_NORMAL)

        cv2.imshow('Debug', img)
        # cv2.imwrite(f'Img/Imagem_{i}.jpeg', img)
        i+=1
        k = cv2.waitKey(1)
        if k == 27:
            break
    grabResult.Release()

# Releasing the resource
camera.StopGrabbing()

cv2.destroyAllWindows()
```

I'm getting this error message:

`_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16599)`

closed 2020-01-13 by GuilhermeCarvalho1144

amarnathreddy0201, 2022-06-29:
Hi Sir, I Studied above Comments .

The Below code won't work when I try to set the camera parameters after camera.StartGrabbing() called 

How can I set the parameters of height and width after calling camera.StartGrabbing() 

```
import cv2
from pypylon import pylon

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
converter = pylon.ImageFormatConverter()
camera.Open()
camera.CenterX=False
camera.CenterY=False

# demonstrate some feature access
new_width = camera.Width.GetValue() - camera.Width.GetInc()
if new_width >= camera.Width.GetMin():
    camera.Width.SetValue(new_width)

numberOfImagesToGrab = 100
camera.StartGrabbing()
camera.Open()
print("Max gain",camera.Gain.Max)
print("Min gain",camera.Gain.Min)
print("Max ExposureTime",camera.ExposureTime.Max)
print("Min ExposureTime",camera.ExposureTime.Min)
i=0
j=0
while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    
    if grabResult.GrabSucceeded():
        i+=1
        j+=1
        if i==700:
            print(i)
            #camera.Open()
            camera.Height.SetValue( 200)
            camera.Width.SetValue( 300)
            
            camera.Gain=5
            camera.ExposureTime=8000
            i=0
        if j==500:
            #camera.Open()
            print(j)
            
            camera.Height.SetValue(1088)
            camera.Width.SetValue( 2048)
            
            camera.Gain=20
            camera.ExposureTime=8000
            j=0
        # Access the image data.
        
        print("AMAR :",camera.IsOpen())
        image = converter.Convert(grabResult)
        img = image.GetArray()
        cv2.imshow("image",img)
        cv2.waitKey(1)

grabResult.Release()
camera.Close()
```

thiesmoeller, 2022-06-29:
Please create a new ticket with your request.

amarnathreddy0201, 2022-06-30:
Thank you for you're response Sir 

I Used above snippet for changing the parameters.

Thanks,

ANR .S

oak-tree, 2023-03-30:
for us we had same error on different parameters,  weirdly, it worked for one camera and did not for the same camera model.

but after running pylonviewer, it worked on both camera

[image]