basler/pypylon #195: Error when trying to install pypylon on Windows (32-bit)

alexzauritz, 2020-01-20:
When I am typing `pip install .` inside the pypylon folder, i get the following error code:

`
WARNING:root:git not found or invalid tag found.
WARNING:root:-> Building version from date!
Building version: 2020.1.20.dev52496+pylon5.0.10
Remove: .\generated
Remove: .\pypylon
call C:\miniSpec\Software\swigwin-4.0.1\swig.exe -python -outdir C:\miniSpec\Software\pypylon-master\pypylon -o C:\miniSpec\Software\pypylon-master\generated\genicam_wrap.cpp -c++ -Wextra -Wall -threads -py3 -DHAVE_PYLON_GUI -D_WIN32 -IC:\Program Files\Basler\pylon 5\development\include genicam.i
NodeCallback.i(166) : Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
IPort.i(23) : Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
C:\Program Files\Basler\pylon 5\development\include\GenApi\IBase.h(55) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
PortImpl.i(52) : Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
C:\Program Files\Basler\pylon 5\development\include\GenApi\IBase.h(55) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.

call C:\miniSpec\Software\swigwin-4.0.1\swig.exe -python -outdir C:\miniSpec\Software\pypylon-master\pypylon -o C:\miniSpec\Software\pypylon-master\generated\pylon_wrap.cpp -c++ -Wextra -Wall -threads -py3 -DHAVE_PYLON_GUI -D_WIN32 -IC:\Program Files\Basler\pylon 5\development\include -IC:\Program Files\Basler\pylon 5\development\include pylon.i
C:\Program Files\Basler\pylon 5\development\include\pylon\gige\ActionTriggerConfiguration.h(70) : Warning 325: Nested struct not currently supported (ActionParameter ignored)
..\genicam\NodeCallback.i(166) : Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
..\genicam\IPort.i(23) : Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
C:\Program Files\Basler\pylon 5\development\include\GenApi\IBase.h(55) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
..\genicam\PortImpl.i(52) : Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
C:\Program Files\Basler\pylon 5\development\include\GenApi\IBase.h(55) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
C:\Program Files\Basler\pylon 5\development\include\pylon\PylonImageBase.h(38) : Warning 403: Class 'Pylon::CPylonImageBase' might be abstract, no constructors generated,
C:\Program Files\Basler\pylon 5\development\include\pylon\ReusableImage.h(50) : Warning 403: Method Pylon::IReusableImage::IsSupportedPixelType(Pylon::EPixelType) const might not be implemented.

Copy C:\Program Files\Basler\pylon 5\development\..\runtime\Win32\PylonUsb_MD_VC120_V5_0_TL.dll => C:\miniSpec\Software\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 5\development\..\runtime\Win32\uxapi_v10.dll => C:\miniSpec\Software\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 5\development\..\runtime\Win32\PylonGUI_MD_VC120_v5_0.dll => C:\miniSpec\Software\pypylon-master\pypylon\
Copy C:\Program Files\Basler\pylon 5\development\..\runtime\Win32\PylonUtility_MD_VC120_v5_0.dll => C:\miniSpec\Software\pypylon-master\pypylon\
Copy tree C:\Program Files\Basler\pylon 5\development\..\runtime\Win32\pylonCXP => C:\miniSpec\Software\pypylon-master\pypylon\pylonCXP
Traceback (most recent call last):
  File "setup.py", line 900, in <module>
    bs.copy_runtime()
  File "setup.py", line 481, in copy_runtime
    super(BuildSupportWindows, self).copy_runtime()
  File "setup.py", line 211, in copy_runtime
    shutil.copytree(src, dst)
  File "C:\Users\Jo\AppData\Local\Programs\Python\Python38-32\lib\shutil.py", line 546, in copytree
    with os.scandir(src) as itr:
FileNotFoundError: [WinError 3] System can not found the path: 'C:\\Program Files\\Basler\\pylon 5\\development\\..\\runtime\\Win32\\pylonCXP'
`

Is there any way to fix it?

thiesmoeller, 2020-01-20:
On Windows please install pylon 6.

Check that that PYLON_ROOT is not pointing to old pylon 5

alexzauritz, 2020-01-20:
thanks a lot

alexzauritz, 2020-01-21:
What is PYLON_ROOT and where to find it? I got still the same error message after installing pylon 6

alexzauritz, 2020-01-21:
After uninstalling pylon5 (and the full directory) I installed pylon 6 but i still get a very similar error message.

```
 ERROR: Command errored out with exit status 1:
     command: 'c:\users\jo\appdata\local\programs\python\python38-32\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Jo\\AppData\\Local\\Temp\\pip-req-build-blt508l3\\setup.py'"'"'; __file__='"'"'C:\\Users\\Jo\\AppData\\Local\\Temp\\pip-req-build-blt508l3\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\pip-egg-info'
         cwd: C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\
    Complete output (38 lines):
    WARNING:root:git not found or invalid tag found.
    WARNING:root:-> Building version from date!
    NodeCallback.i(166) : Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
    IPort.i(23) : Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
    C:\Program Files\Basler\pylon 5\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
    PortImpl.i(52) : Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
    C:\Program Files\Basler\pylon 5\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
    C:\Program Files\Basler\pylon 5\Development\include\pylon\gige\ActionTriggerConfiguration.h(71) : Warning 325: Nested struct not currently supported (ActionParameter ignored)
    ..\genicam\NodeCallback.i(166) : Warning 401: Nothing known about base class 'CNodeCallback'. Ignored.
    ..\genicam\IPort.i(23) : Warning 403: Class 'GENAPI_NAMESPACE::IPort' might be abstract, no constructors generated,
    C:\Program Files\Basler\pylon 5\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
    ..\genicam\PortImpl.i(52) : Warning 403: Class 'GENAPI_NAMESPACE::CPortImpl' might be abstract, no constructors generated,
    C:\Program Files\Basler\pylon 5\Development\include\GenApi\IBase.h(57) : Warning 403: Method GENAPI_NAMESPACE::IBase::GetAccessMode() const might not be implemented.
    C:\Program Files\Basler\pylon 5\Development\include\pylon\PylonImageBase.h(38) : Warning 403: Class 'Pylon::CPylonImageBase' might be abstract, no constructors generated,
    C:\Program Files\Basler\pylon 5\Development\include\pylon\ReusableImage.h(50) : Warning 403: Method Pylon::IReusableImage::IsSupportedPixelType(Pylon::EPixelType) const might not be implemented.
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\setup.py", line 900, in <module>
        bs.copy_runtime()
      File "C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\setup.py", line 481, in copy_runtime
        super(BuildSupportWindows, self).copy_runtime()
      File "C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\setup.py", line 211, in copy_runtime
        shutil.copytree(src, dst)
      File "c:\users\jo\appdata\local\programs\python\python38-32\lib\shutil.py", line 546, in copytree
        with os.scandir(src) as itr:
    FileNotFoundError: [WinError 3] System can not find path: 'C:\\Program Files\\Basler\\pylon 5\\Development\\..\\runtime\\Win32\\pylonCXP'
    Building version: 2020.1.21.dev28450
    Remove: .\generated
    Remove: .\pypylon
    call C:\miniSpec\Software\swigwin-4.0.1\swig.exe -python -outdir C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\pypylon -o C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\generated\genicam_wrap.cpp -c++ -Wextra -Wall -threads -py3 -DHAVE_PYLON_GUI -D_WIN32 -IC:\Program Files\Basler\pylon 5\Development\include genicam.i

    call C:\miniSpec\Software\swigwin-4.0.1\swig.exe -python -outdir C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\pypylon -o C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\generated\pylon_wrap.cpp -c++ -Wextra -Wall -threads -py3 -DHAVE_PYLON_GUI -D_WIN32 -IC:\Program Files\Basler\pylon 5\Development\include -IC:\Program Files\Basler\pylon 5\Development\include pylon.i

    Copy C:\Program Files\Basler\pylon 5\Development\..\runtime\Win32\PylonGUI_v6_0.dll => C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\pypylon\
    Copy C:\Program Files\Basler\pylon 5\Development\..\runtime\Win32\PylonUtility_v6_0.dll => C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\pypylon\
    Copy tree C:\Program Files\Basler\pylon 5\Development\..\runtime\Win32\pylonCXP => C:\Users\Jo\AppData\Local\Temp\pip-req-build-blt508l3\pypylon\pylonCXP
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

```

thiesmoeller, 2020-01-21:
You have to install all transportlayers from the pylon installer.

The setup.py assumes that you want to create a generic pypylon and requires a full install of pylon

alexzauritz, 2020-01-22:
But why did the .exe not install everything I need? And how to install all transportlayers?

MCilento93, 2020-07-27:
is there any differences in run `pip3 install pypylon-1.5.4-cp34-cp34m-win_amd64.wh`l of `pip3 install pypylon-1.5.4-cp35-cp35m-win_amd64.whl` or the other released?

thiesmoeller, 2020-07-27:
cp stands for CPython

You have to install the version that matches your interpreter

thiesmoeller, 2020-07-27:
@alexzauritz 
   In the install dialogue choose custom. Then you can select all transportlayers

thiesmoeller, 2021-02-23:
Beginning with release 1.7.1 we have binary releases on all platforms, that you can install via pip.

update your pip version before by calling `python -m pip install --upgrade pip`

closed 2021-02-23 by thiesmoeller