basler/pypylon #197: Errors in running samples

vishwas-rayne, 2020-02-03:
Hi,

I am trying to get a basler camera working inside a docker container without setting network mode to host.
I tried running some of the samples provided in the samples directory.
The ['grab.py'](https://github.com/basler/pypylon/blob/master/samples/grab.py) sample works fine but am running into issues when i try to run the ['gige_ip_config.py'](https://github.com/basler/pypylon/blob/master/samples/gige_announce_renounce.py) & ['gige_announce_renounce.py'](https://github.com/basler/pypylon/blob/master/samples/gige_ip_config.py) samples, i run into the following error:

### Traceback (most recent call last):
  File "gige_announce_renounce.py", line 14, in <module>
    cam_info.GetIpAddress(),
  File "/usr/local/lib/python3.6/dist-packages/pypylon/pylon.py", line 655, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, DeviceInfo, name)
  File "/usr/local/lib/python3.6/dist-packages/pypylon/pylon.py", line 108, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'DeviceInfo' object has no attribute 'GetIpAddress'

Following are the pre-requisites' versions:
1. Pylon - pylon-5.1.0.12682
2. Swig - SWIG Version 3.0.12
3. Python - Python 3.6.9

thiesmoeller, 2020-02-03:
What is your pypylon version?

Built from master or using 1.5.1 from releases ?

vishwas-rayne, 2020-02-03:
I built from master which is actually the latest 1.5.1 release. Both the commit ids match(409c5e09a175f469bd82ecb016619fffeeb351c5).

I have replicated my issue using a docker container here at (https://github.com/vishwas-rayne/pypylondocker). I am able to reproduce the issue by running the samples from within docker containers running with the same pre-requisites.

thiesmoeller, 2020-02-03:
please check, if you really run pypylon 1.5.1 

`python3 -c 'from pypylon import pylon as py ; print(py.__version__)'`

function DeviceInfo.GetIpAddress is available only in >= 1.5.1

vishwas-rayne, 2020-02-04:
I got this as the pypylon version when i ran the above command:
`1.5.1+pylon5.1.0`

I also checked the source code present in (/usr/local/lib/python3.6/dist-packages/pypylon/pylon.py). I see a few functions available like GetSerialNumber & GetModelName but not the GetIpAddress for the DeviceInfo class.

thiesmoeller, 2020-02-04:
for the GEV features to work you need the latest linux pylon  -> 5.2
https://www.baslerweb.com/en/sales-support/downloads/software-downloads/pylon-5-2-0-linux-x86-64-bit/

vishwas-rayne, 2020-02-05:
Samples work fine after updating Pylon to 5.2.0

closed 2020-02-05 by vishwas-rayne