basler/pypylon #198: TimeoutException Error

GuilhermeCarvalho1144, 2020-02-03:
I'm getting this error whem the camera is rolling for a while:

> File "site-packages\pypylon\pylon.py", line 3222, in RetrieveResult
_genicam.TimeoutException: Grab timed out. The acquisition is not started. : TimeoutException thrown (file 'instantcameraimpl.h', line 1031)

The code I'm using is the following:

```
import datetime
from pypylon import pylon
import numpy as np
import cv2

TAMANHO_X = 4096
TAMANHO_Y = 7053

tl_factory = pylon.TlFactory.GetInstance()
# conecting to the first available camera
camera = None
for dev_info in tl_factory.EnumerateDevices():
    if dev_info.GetDeviceClass() == "BaslerGigE":
        print("using %s @ %s" % (dev_info.GetModelName(), dev_info.GetIpAddress()))
        camera = pylon.InstantCamera(tl_factory.CreateDevice(dev_info))
        break
else:
    raise EnvironmentError("no GigE device found")

camera.Open()
camera.Width.SetValue(TAMANHO_X)
camera.Height.SetValue(TAMANHO_Y)
camera.OffsetX.SetValue(0)
camera.OffsetY.SetValue(0)

camera.TriggerSelector.SetValue("FrameStart")
camera.TriggerMode.SetValue("On")
camera.TriggerSource.SetValue("Line1")
camera.TriggerActivation.SetValue("RisingEdge")
# camera.AcquisitionMode.SetValue("SingleFrame")
camera.ExposureTimeAbs.SetValue(250)

# Grabing Continusely (video) with minimal delay
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
i = 0
imagens = []
imagens_concatenadas= []
while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(500000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data
        time1 = datetime.datetime.now()
        image = converter.Convert(grabResult)
        img = image.GetArray()
        cv2.namedWindow("Debug", cv2.WINDOW_NORMAL)

        cv2.imshow("Debug", img)
        img_gray = cv2.cvtColor(img, cv2.COLOR_BGR2GRAY)
        img_zeros = np.zeros((TAMANHO_Y, TAMANHO_X), dtype=np.uint8)
        if (img_gray-img_zeros).all() != 0:
            cv2.imwrite(f"Img/Imagem_{i}.jpeg", img_gray)
            i += 1
            
        k = cv2.waitKey(1)
        if k == 27:
            break

        img_gray_rotate = cv2.rotate(img_gray, cv2.ROTATE_90_CLOCKWISE)
        imagens.append(img_gray_rotate)
        if(len(imagens) == 7):
            imagens.reverse()
            imagens_concatenadas = np.concatenate(imagens, axis=1)
            print(f"Tempo de processamento da imagem ate agora {datetime.datetime.now()-time1}")
            imagens = []
        
    grabResult.Release()

# Releasing the resource
camera.StopGrabbing()

cv2.destroyAllWindows()
```

The camera I'm using is the Basler Racer raL4096-24gm.

PolymaTh-EE98, 2020-04-10:
You are using the hardware trigger.  Is it sitting idle? This will cause a timeout.

MCilento93, 2020-12-27:
> 500000

Does this code can be used for grab a simple photo? I mean leaving the main while loop as soon as you get a succeded image?

```python
import numpy as np
NULL_ARRAY=np.ndarray(shape=(0,0))

def grab_basler_photo():
    array=NULL_ARRAY
    try:
        from pypylon import pylon
        camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
        camera.Open()
        
        # Grabing Continusely (video) with minimal delay
        camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
        converter = pylon.ImageFormatConverter()
        converter.OutputPixelFormat = pylon.PixelType_BGR8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
        
        while camera.IsGrabbing():
            grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
            if grabResult.GrabSucceeded():
                image = converter.Convert(grabResult)
                array = image.GetArray()
                camera.Close()
                return array
            else:
                pass
            grabResult.Release()
        camera.StopGrabbing()
        camera.Close()
        
    except Exception as Err:
        array=NULL_ARRAY
        logger.error('errors with pypylon\n'+Err)
        
    return array
```

closed 2025-12-15 by thiesmoeller