basler/pypylon #199: How to use the GPIOs of the camera Basler Racer raL4096-24gm

GuilhermeCarvalho1144, 2020-02-03:
I´m trying check the status of input using pypylon, but I'm getting this message.

> _genicam.InvalidArgumentException: Feature 'LineSelector' : cannot convert value 'LineSelector_Line3', the value is invalid. : InvalidArgumentException thrown in node 'LineSelector' while calling 'LineSelector.FromString()' (file 'enumeration.cpp', line 134)

Im trying to access the GPIO based on the code i have that uses pylon  C++ .
The code is the following:

```
import datetime
from pypylon import pylon
import numpy as np
import cv2

TAMANHO_X = 4096
TAMANHO_Y = 7053

tl_factory = pylon.TlFactory.GetInstance()
# conecting to the first available camera
camera = None
for dev_info in tl_factory.EnumerateDevices():
    if dev_info.GetDeviceClass() == "BaslerGigE":
        print("using %s @ %s" % (dev_info.GetModelName(), dev_info.GetIpAddress()))
        camera = pylon.InstantCamera(tl_factory.CreateDevice(dev_info))
        break
else:
    raise EnvironmentError("no GigE device found")

camera.Open()
camera.Width.SetValue(TAMANHO_X)
camera.Height.SetValue(TAMANHO_Y)
camera.OffsetX.SetValue(0)
camera.OffsetY.SetValue(0)

camera.TriggerSelector.SetValue("FrameStart")
camera.TriggerMode.SetValue("On")
camera.TriggerSource.SetValue("Line1")
camera.TriggerActivation.SetValue("RisingEdge")
# camera.AcquisitionMode.SetValue("SingleFrame")
camera.ExposureTimeAbs.SetValue(250)

camera.LineSelector.SetValue("LineSelector_Line3")
camera.LineMode.SetValue("LineMode_Input")
enable = camera.LineStatus.GetValue()

# Grabing Continusely (video) with minimal delay
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()
```

thiesmoeller, 2020-02-03:
It's not "LineSelector_Line3" but "Line3"
... Would be the same in c++ if you use the string argument.

Same for some more code lines you have

GuilhermeCarvalho1144, 2020-02-06:
thanks, it worked.
Closing the issue.

closed 2020-02-06 by GuilhermeCarvalho1144