basler/pypylon #200: Can't grab by using a camera from different subnet

vishwas-rayne, 2020-02-05:
Am using the latest pylon & pypylon.
I have a simple code snippet to add camera belonging to a different subnet into the current subnet and to start grabbing:
`   factory = pylon.TlFactory.GetInstance()
    ptl = factory.CreateTl('BaslerGigE')

    empty_camera_info = ptl.CreateDeviceInfo()
    #ip = 'SomeIpAddress'
    empty_camera_info.SetPropertyValue('IpAddress', ip)

    camera_device = factory.CreateDevice(empty_camera_info)
    cam_obj = pylon.InstantCamera(camera_device)
    grab(cam_obj)`
The grab(cam_obj) function mimics the functionality of the [grab.py](https://github.com/basler/pypylon/blob/master/samples/grab.py) sample from the pypylon samples.

Am able to get the print `Using device  acA1920-40gc` meaning adding the camera to the subnet was indeed possible.
However, am running into the below error during the camera.RetrieveResult() function is being executed:
`Traceback (most recent call last):
  File "basler_three.py", line 38, in grab
    grabResult = camera.RetrieveResult(50000, pylon.TimeoutHandling_ThrowException)
  File "/usr/local/lib/python3.6/dist-packages/pypylon/pylon.py", line 3588, in RetrieveResult
    return _pylon.InstantCamera_RetrieveResult(self, *args)
_genicam.TimeoutException: Grab timed out. The acquisition is not started. : TimeoutException thrown (file 'InstantCameraImpl.h', line 1031)`

I tried to replicate the same using the PylonViewerApp and i faced the same scenario.
Am able to add the camera using 'Add Remote GigE Camera' option and it is visible in the devices tab but unable to get any images in single or continuous shot.

Is the option to grab images by adding a camera from a different subnet not supported as of latest Pylon or some extra steps need to be ensured to get this working ?

GuilhermeCarvalho1144, 2020-02-10:
According to this document that I found on the basler web site [link here](https://www.baslerweb.com/en/sales-support/knowledge-base/frequently-asked-questions/i-ve-installed-the-basler-pylon-viewer-and-a-pylon-network-driver-either-the-basler-filter-driver-or-the-basler-performance-driver-but-i-do-not-see-my-camera-in-the-view-tree-of-the-pylon-viewer-why-not/15183/). 

You cannot access cameras from a different subnet.

lcflorindo, 2020-07-22:
I have the same situation in here, but I am able to find my device either by code or Pylon Viewer, even in different subnets.
My true issue is that it took 3.2 seconds to grab by code but real time by PylonViewer...
I do not understand where the restrition is located and how to solve by code. Any ideas?

kobotschick, 2020-07-25:
I have the same issue within docker container. 
Running on windows docker desktop. The camera can be found , but not able to retrieve images. It’s failed with timeout. It’s work‘s with