basler/pypylon #201: Cannot retrieve RGB frame from harware triggered camera ?

ParadoxRobotics, 2020-02-05:
Hi, 
I am currently working on a industrial project requiring hardware trigger on a Basler Aca4600-10uc USB3.0 camera. I write a simple program able to retrieve image from the camera in this kind of mode. (The hardware trigger is done every 2s on the rising edge)
Unfortunately, the following code can only retrieve frame on a grayscale level; is it possible to get the RGB one ?

Thanks in advance !

`
# implementation of a trigger based acquition method for basler camera
# based on pypylon library
# Author:  Munch Quentin, 2020,
from pypylon import pylon
import cv2

# camera init
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
# get access_stream
camera.Open()

camera.AcquisitionStatusSelector = "FrameTriggerWait"
camera.TriggerSelector.SetValue('FrameStart')
camera.TriggerSource.SetValue("Line1")
camera.TriggerMode.SetValue("On")
camera.TriggerActivation.SetValue('RisingEdge')
camera.PixelFormat.SetValue('Mono8')
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

# image conversion
converter = pylon.ImageFormatConverter()
# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

while True:
    # wait for grab
    grabResult = camera.RetrieveResult(20000, pylon.TimeoutHandling_ThrowException)
    # if grab succesfull
    if grabResult.GrabSucceeded():
        image = converter.Convert(grabResult)
        img = grabResult.GetArray()
        img = cv2.resize(img, (640, 480))
        cv2.namedWindow('title', cv2.WINDOW_NORMAL)
        cv2.imshow('title', img)
        k = cv2.waitKey(1)
        if k == 27:
            break
    grabResult.Release()

# Releasing the resource
camera.StopGrabbing()
cv2.destroyAllWindows()
`

thiesmoeller, 2020-02-05:
.. please use the correct markdown to post code .. makes it more readable. Just select your code and click the 'code' button above the edit window.

You are using the wrong image in your code.
```
image = converter.Convert(grabResult)
img = grabResult.GetArray()
```

It has to be

```
image = converter.Convert(grabResult)
img = image.GetArray()
```

And please select a yuv format before start grabbing.

`camera.PixelFormat = "YCbCr422_8`

The device wakes up in a Bayer raw format. If you leave it there you have to do the Bayer conversion to rgb on your PC. 
By switching the Pixelformat to yuv you will get already color information. The conversion to rgb in the imageformat converter is cheap in this case

ParadoxRobotics, 2020-02-06:
Thank for your help,
I am sorry, i sent the code with my phone...
Thanks also for the tip for the image conversion and the correction for the mistake I made in my code.

closed 2020-12-01 by thiesmoeller