basler/pypylon #203: Getting an error while trying compiling from the source code

saikatpph, 2020-02-12:
I am using Python 3.8, and would like to install the pypylon. Since there is no binding available for the Python 3.8 version, I was trying to compile it from the source code, following the instruction:
     " git clone https://github.com/basler/pypylon.git
       cd pypylon
       pip install . "
However, I came across the following error. 
 Any suggestion to fix this issue?
Thanks

[Error output.txt](https://github.com/basler/pypylon/files/4189586/Error.output.txt)

thiesmoeller, 2020-02-12:
Reinstall Pylon 6 and Install all transportlayers including cxp.

saikatpph, 2020-02-12:
> Reinstall Pylon 6 and Install all transportlayers including cxp.

Thanks a lot !!! it Worked. :)

closed 2020-02-12 by saikatpph