basler/pypylon #205: GrabOne.py sample Error of genicam.LogicalErrorException: Cannot create chunk parser. No chunk node map description available. Check the Transport Layer settings. : LogicalErrorException thrown (file 'ChunkParsingEngine.h', line 116)

HajimeMiyata, 2020-02-18:
Hello,

We try to grab image by grabone.py sample following URL.
https://github.com/basler/pypylon/blob/5cd49e44d719eb7d122a86d78a6f2c95ec33881b/samples/grabone.py

The source code cause following error.
Traceback (most recent call last):
  File "grabone.py", line 20, in <module>
    result = camera.GrabOne(100)
  File "/usr/local/share/work/plib/pypylon/pylon.py", line 3343, in GrabOne
    return _pylon.InstantCamera_GrabOne(self, *args)
_genicam.LogicalErrorException: Cannot create chunk parser. No chunk node map description available. Check the Transport Layer settings. : LogicalErrorException thrown (file 'ChunkParsingEngine.h', line 116)
root@armadillo:/usr/local/share/work# 

We use acA3088-16gc, and Armadillo640(ARM CPU).

It is no problem that this camera connected to Window PC and open by pylon Viewer.
However, it is failed to grab by grabone.py.
It is also no problem to use RetreiveResult C sample whichi is C:\Program Files\Basler\pylon 5\Development\Samples\C\OverlappedGrab\OverlappedGrab.c

thiesmoeller, 2020-02-18:
As a quick trip:
If you don't require chunk data access, don't use it.

The grab one sample demonstrate chunk data access.

Please refer to e.g the sample from the main readme

thiesmoeller, 2025-12-15:
Closing as answered. The chunk parser error occurs when chunk data access is requested but not available. If you don't require chunk data, don't use GrabOne() which enables chunk access. Use the basic grab.py sample instead, or disable chunk features. The GrabOne sample demonstrates chunk data access, but it's not required for basic image grabbing.