basler/pypylon #206: BCON Camera (dart) - example

L-o-r-e-n-z, 2020-02-20:
I try to communicate with the pypylon on an iMX8 with a camera with BCON connection. Unfortunately all the samples do not work and create an exception:
```
root@imx8mmevk-basler:~/camera# python3 grabone.py
Traceback (most recent call last):
  File "grabone.py", line 6, in <module>
    pylon.TlFactory.GetInstance().CreateFirstDevice())
  File "/usr/lib/python3.5/site-packages/pypylon/pylon.py", line 1755, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 692)
root@imx8mmevk-basler:~/camera#
```
When I start the pylon Viewer I see the camera and I can capture images... What do I do wrong?

L-o-r-e-n-z, 2020-02-20:
I made one step forward, now I get a different error message, but still no image can be received:
`root@imx8mmevk-basler:~/camera# GENICAM_GENTL64_PATH=/opt/dart-bcon-mipi/lib python3 grabone.py
20.02.2020 15:32:13.364550 Didn't find a matching capture device for 'daA2500-60mci'
20.02.2020 15:32:13.365420 Didn't find a matching capture device for 'daA2500-60mci'
Traceback (most recent call last):
  File "grabone.py", line 11, in <module>
    camera.ChunkModeActive = True
  File "/usr/lib/python3.5/site-packages/pypylon/pylon.py", line 4562, in __setattr__
    self.GetNodeMap().GetNode(attribute).SetValue(val)
  File "/usr/lib/python3.5/site-packages/pypylon/genicam.py", line 1552, in GetNode
    return _genicam.INodeMap_GetNode(self, Name)
_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16590)
`

thiesmoeller, 2020-02-20:
Please use the sample code from the project readme as first start.
The error message gives you the hint what is not working.

Grabone.py is using chunks, that are not yet available on a BCON for Mipi Interface

L-o-r-e-n-z, 2020-02-21:
thank you very much, now I get some image data:
```
root@imx8mmevk-basler:~/camera# GENICAM_GENTL64_PATH=/opt/dart-bcon-mipi/lib python3 example.py
21.02.2020 06:50:08.117118 Didn't find a matching capture device for 'daA2500-60mci'
21.02.2020 06:50:08.117946 Didn't find a matching capture device for 'daA2500-60mci'
SizeX:  4208
SizeY:  3120
Gray value of first pixel:  [126  11]
```

closed 2020-02-21 by L-o-r-e-n-z