basler/pypylon #207: sample grabmultiple is not useful

madhusudangr, 2020-02-26:
Each time I run the sample I get completely different order of camera trigger, I cant understand a way utilize this sample and create a code to capture frames from different cameras. Can someone help me with the same ?

Example run 1: 
```Using device  acA1300-60gc 21970087
Using device  acA1300-60gc 21970091
Using device  acA1300-60gc 22605500
Camera  0 :  acA1300-60gc 21970087
...
Camera  0 :  acA1300-60gc 21970087
...
Camera  1 :  acA1300-60gc 21970091
...
Camera  0 :  acA1300-60gc 21970087
...
Camera  1 :  acA1300-60gc 21970091
...
Camera  0 :  acA1300-60gc 21970087
...
Camera  1 :  acA1300-60gc 21970091
...
Camera  2 :  acA1300-60gc 22605500
...
Camera  0 :  acA1300-60gc 21970087
...
Camera  1 :  acA1300-60gc 21970091
...
```

Example run 2: 
```
Using device  acA1300-60gc 21970087
Using device  acA1300-60gc 21970091
Using device  acA1300-60gc 22605500
Camera  0 :  acA1300-60gc 21970087
...
Camera  1 :  acA1300-60gc 21970091
...
Camera  0 :  acA1300-60gc 21970087
..
Camera  0 :  acA1300-60gc 21970087
...
Camera  1 :  acA1300-60gc 21970091
..
Camera  0 :  acA1300-60gc 21970087
..
Camera  1 :  acA1300-60gc 21970091
...
Camera  0 :  acA1300-60gc 21970087
...
Camera  2 :  acA1300-60gc 22605500
...
Camera  1 :  acA1300-60gc 21970091
...
```
Am I triggering the camera in a different way ? Is there a better way to trigger the cameras in a fixed sequence ? 

I am using the code grabmultipmle.py from the samples folder.

Curmen, 2020-03-06:
How many cameras you are using ?

MattsProjects, 2020-03-06:
Hi there. The order in which devices are discovered is never guaranteed because some devices may take longer than others to respond. This is especially the case with GigE, as network latencies, etc. can vary. This is also the case for grabbing images. They may arrive in different order when using multiple cameras.
If you absolutley need an InstantCamera object to use only a specific camera, then the way to go is to create a CDeviceInfo object containing such info as a serial number, and then pass this object via the instant camera's Attach call.
If you are simply looking to identify which image came from which camera, use SetCameraContext() to give each instant camera an ID number. This context can then be read from each grab result to know the camera it came from.

closed 2020-12-01 by thiesmoeller