basler/pypylon #209: unable to install/import/use pypylon

thomaswenkin, 2020-03-09:
Hi every one
I am trying to develop an application on Python with pypylon and a Gige Camera from Basler but I am unable to install pypylon using pip even though every requirement is met.

When I try to use pip to install it, I get this error message :
[image]

Then if I try to directly unzip the wheel inside my site-packages folder, pip registers it and all like so :
[image]

And after that, trying to import the package fails even though the package is seen as available :
[image]
It seems that the _pylon directory is missing, as well as a dll file (which I assume is available inside my Pylon Viewer Installation directory)

I already installed Pylon Viewer twice, each time making sure that the SDK installing box was checked to be sure to meet the requirements.

Does anyone have a hint or something else to help me troubleshoot this?

Thanks in Advance

thiesmoeller, 2020-03-09:
From the error message you try to load a module from a wrong CPU architecture or os.
..."is not a valid Win32 app...."

What is your os/system?
What whl file you tried to install?

thomaswenkin, 2020-03-12:
I am feeling kinda dumb, my Python install was using Python 3.8 while Pypylon hasn't been ported to 3.8 yet...
I installed Python 3.7 and everything's fine now

closed 2020-03-12 by thomaswenkin