basler/pypylon #216: BCON: change exposure time

L-o-r-e-n-z, 2020-03-25:
Hello all

how can I change the exposure time on a BCON camera? Is this feature not yet implemented? Is there a list of functions which shows the implemented functions of pypylon for BCON cameras?
The following gives an exception:
`
camera.ExposureTimeAbs = 20000
`

`
    self.camera.ExposureTimeAbs = 20000
  File "/usr/lib/python3.5/site-packages/pypylon/pylon.py", line 4562, in __setattr__
    self.GetNodeMap().GetNode(attribute).SetValue(val)
  File "/usr/lib/python3.5/site-packages/pypylon/genicam.py", line 1552, in GetNode
    return _genicam.INodeMap_GetNode(self, Name)
_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16590)
`

Thank you for your support
Lorenz

L-o-r-e-n-z, 2020-03-25:
I just saw that there is a new release of the i.MX8 package which seems to support the following:
     - New manual gain feature.
     - New manual exposure time.

I will try it the next days.

jhacsonmeza, 2020-04-08:
Hello, have you tried with `camera.ExposureTime.SetValue(20000)`?

L-o-r-e-n-z, 2020-04-08:
thank you, yes it is working with the following lines:
`camera.ExposureTime.SetValue(1000.0)
camera.ExposureAuto.SetValue("Off")`

closed 2020-04-08 by L-o-r-e-n-z

chenyuyi94, 2020-08-24:
> Hello all
>
> how can I change the exposure time on a BCON camera? Is this feature not yet implemented? Is there a list of functions which shows the implemented functions of pypylon for BCON cameras? [...]
hi sir:
May i ask you which verion pypylon and pylon you had used.
I have used camera.Open() before setting the exposure time.
I can set camera exposuretime, using 'camera.ExposureTimeAbs = 20000',but whern i using 'camera.ExposureTime.SetValue(1000.0)'.

thiesmoeller, 2020-08-24:
@chenyuyi94 
You are using a gigabit Ethernet camera.
This thread was about BCON ( lvds and mipi cameras)

The interface of the device is defined by the camera not pylon.
Please check with pylon viewer or docs.baslerweb.com for the correct feature names.

Feature names in C++ are identical to python !

chenyuyi94, 2020-08-24:
I get it，thank you for your unswer。

 祝您天天好心情！