basler/pypylon #217: No module named '_pylon'

juliomilani, 2020-04-06:
I tried installing pypylon in different ways (through Anaconda Python 3.7 / Native Python 3.6 / From wheel...) but I always get the same error.
I already successfully installed pylon and swig. When I build from source, it installs just fine but i get the following error when trying to "from pypylon import pylon":

Traceback (most recent call last):
  File "/home/julio/pypylon/pypylon/pylon.py", line 42, in swig_import_helper
    return importlib.import_module(mname)
  File "/home/julio/anaconda3/envs/tf-gpu/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'pypylon._pylon'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/julio/pypylon/pypylon/pylon.py", line 45, in <module>
    _pylon = swig_import_helper()
  File "/home/julio/pypylon/pypylon/pylon.py", line 44, in swig_import_helper
    return importlib.import_module('_pylon')
  File "/home/julio/anaconda3/envs/tf-gpu/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_pylon'

Specs:
Ubuntu 18.04
Python 3.7.6
Anaconda 3

Can you help me please.

juliomilani, 2020-04-06:
I solved the issue by uninstalling pylon 6.1.0 and installing pylon 5.2.

closed 2020-04-06 by juliomilani

aquilesC, 2020-12-22:
You can go back to pylon6, but you have to edit the setup file (see #308 ) and then run ``python setup.py install``