basler/pypylon #226: Can PyPylon intercept mouse click coordinates?

bwllc, 2020-04-29:
Hello everyone,

I have been asked to reproduce a custom application that another engineer on our team saw once, but which we do not have available in-house.  Basler cameras were used for vision in this application.  An object on a movable stage was being observed.  The user could click on a live image window.  The application intercepted the mouse click coordinate, and knew the coordinates of the live image window.  From this information, motors were commanded to move the clicked point to the center of the camera field.  MATLAB was apparently required in the software stack.

I've done a little research, and it's possible that this was accomplished using Pylon Viewer, exporting data to MATLAB through the GenTL protocol.  This application didn't need to do any image processing or storage, it just needed to display images, so that the operator could choose a pixel to click.  Can MATLAB talk back to the camera, to adjust the camera gain or the exposure time?   I think you would still have to do that by opening Pylon Viewer.

This all may have worked, but it seems very cumbersome to me.

If there is a way for me to capture mouse information using a Pylon Viewer API such as PyPylon, this would be very helpful to me.  I hope you will tell me that this is possible!  If it is not, please suggest an efficient way forward.  I program regularly in Python and in C.  My employer wants this application to work on a Windows 64-bit platform.

Thanks!

PolymaTh-EE98, 2020-04-29:
You can accomplish this with OpenCV mouse click events.

bwllc, 2020-05-01:
> You can accomplish this with OpenCV mouse click events.

Thanks, I'll look into that approach.

stefanklug, 2020-08-28:
Closing as there was no activity in the last month.

closed 2020-08-28 by stefanklug