basler/pypylon #227: Best practices for continuous video (GrabOne vs. StartGrabbingMax)?

bwllc, 2020-04-30:
I've managed to put together the skeleton of a PyPylon application in just a couple of days.  I appreciate the development work that went in to this tool!

My goal is continuous video with a GUI.  I don't need to record anything, just watch a live screen.  The number of images I acquire is theoretically infinite.  I will run camera data acquisition in an independent thread or process from my GUI.

I got camera.GrabOne() to work.  I could just call this forever in a Python loop.  I also got camera.StartGrabbingMax() to work, but you supply a maximum number of images to this method -- and as I said, there is no maximum.  Is there a way to tell StartGrabbingMax() to run indefinitely?  If yes: is there any advantage to doing things that way?  Are there any other methods I should be studying?

Thanks for your help.

thiesmoeller, 2020-05-02:
Look into the grab strategies sample https://github.com/basler/pypylon/blob/a3ca9123979268ddc6ae986cb35f75f6523f5481/samples/grabstrategies.py#L45

StartGrabbing will run indefinitely.
StopGrabbing will end it.

bwllc, 2020-05-03:
Thanks @thiesmoeller, I see that there are some good comments included in that program to help understand what is happening.

stefanklug, 2020-08-28:
This seems solved.

closed 2020-08-28 by stefanklug