basler/pypylon #230: Delay using multiple cameras

CobeA527, 2020-05-08:
Hi Basler - team, 
I am using 3 Basler GigE acA640-120gm on linux. The cameras are connected via a switch with my Laptop. 
In the beginning when  I was trying to continuously grab images from all 3 cameras the Error 3774873620 occured: 

> "The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size."

Therefore I decided to set all parameters in the pylon viewer (Inter_Packet Delay = 5000 & Packet Size = 2000 (max)) and changed and set my Ethernet connection to use jumboframes with MTU = 9000. I also limited each cameras bandwidth to 8Mb/s and changed the etc/security/limits.conf file to ensure real-time priority. 

After that the Error doesn't occurred anymore but the the image acquisition has a huge delay and the video seems like "slow motion". 

Is there anything I can do that I get real-time images/video for every camera ? 
Do i need different settings or is it a hardware problem ? 

Switch: http://www.cudytech.com/productinfo/1451013.html
Ethernet cable: https://www.amazon.de/gp/product/B07RTCZN6G/ref=ppx_yo_dt_b_asin_image_o03_s00?ie=UTF8&psc=1
Ethernet adapter: Intel(R) PRO/1000 Network Connection

Thanks for your help!

thiesmoeller, 2020-05-11:
you are using a 100MBit switch.
as  a single acA640-120gm will produce, even in Mono8, 35MB per sec this can't work.

The issue that you see is most probably the following:
The switch is not able to transfer the data from all three cameras to your PC. Typically a switch will now issue PAUSE frames to the cameras, to force them to pause the video transfer.
As every camera has enough memory to buffer a larger number of VGA frames, you will get them now one by one.

What you observe is the typical behavior of an acquisition and transmission bandwidth mismatch.

Solutions:
* Best would be to use a 1Gbit/s switch like https://www.baslerweb.com/de/produkte/vision-komponenten/netzwerk-und-peripheriegeraete/gigabit-ethernet-smart-switch-gs110tp-poe-8-ports/

CobeA527, 2020-05-15:
Ok, thanks for the fast reply @thiesmoeller ! Is there an option to turn down the acquisition bandwidth for now? The speed of the network card is also 100Mbit/s..

thiesmoeller, 2020-05-15:
Please use the bandwidth manager from pylon viewer to find optimal settings for your setup.
These have to be applied in your pypylon program afterwards! 

https://docs.baslerweb.com/bandwidth-manager.html

TechnoSwiss, 2020-05-28:
Was debating between opening a new issue and adding to this one, but given the error message is identical I'll go ahead and put my question here.

I'm running into this same error trying to run the grab.py demo code. I know values I need to set Inter-Packet Delay and Frame Transmission Delay to (and Brandwidth Reservation) based on testing with Bandwidth-Manager and pylon Viewer, however I haven't been able to find documentation or examples on how to set those values in pypylon, am I missing something? Thanks.

thiesmoeller, 2020-05-29:
For every feature you control in pylon viewer there is a c++ help in the documentatio window of pylon viewer.

Python is nearly 1:1to c++.

camera.Gain = 42;

camera.Gain = 42

...
The controls that the bandwidth tool optimizes are in the 'TransportLayer' section

TechnoSwiss, 2020-05-29:
That did the trick, thanks!

thiesmoeller, 2025-12-15:
Closing as answered. The delay is caused by using a 100Mbit switch which cannot handle the bandwidth from 3 GigE cameras (each producing ~35MB/s in Mono8). The switch issues PAUSE frames, causing the "slow motion" effect. Solution: Use a 1Gbit switch. Also use the Bandwidth Manager tool in Pylon Viewer to optimize settings, then apply those settings in your pypylon code via the TransportLayer parameters.

closed 2025-12-15 by thiesmoeller