basler/pypylon #233: If I want to use camera in hardware triggering mode ,which program I can reference?

shenjun1994, 2020-05-09:
Now, I want to use balser camera in hardware triggering mode, which program I can reference to use?

crunchyluke, 2020-07-14:
I couldn't find any example to reference for this, but first, set the TriggerMode to "On":
camera.TriggerMode.SetValue("On")

Then, camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) and result = camera.RetrieveResult(TIMEOUT, pylon.TimeoutHandling_ThrowException), as usual. The RetrieveResult method should block until either a trigger is received or the TIMEOUT period elapses and throws an exception.

I don't have a concrete example to provide, but I implemented this a while ago and this is the general procedure I used - hope it gives you a starting point at least.

niallodowd, 2020-10-05:
For some reason, I can't permanently set the trigger mode to "On" via python. It works in pylon viewer, but I tries the below code:

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

camera.Open()
camera.TriggerMode.SetValue("On")
camera.Close()

camera.Open()
print(camera.TriggerMode.Value)
camera.Close()

and it just prints "Off" meaning the camera trigger mode is off. 

Is there a better way to switch the trigger mode?

thiesmoeller, 2020-12-01:
camera state is not persisted for genicam based cameras like the Basler cameras.

You habe to explicitly save the data. check https://docs.baslerweb.com/user-sets

closed 2020-12-01 by thiesmoeller