basler/pypylon #234: Question about setting ExposureMode.

kfuna, 2020-05-13:
I want to set the ExposureMode explicitly, but the following error occurs.
What are the possible causes?

Error:
python .\grab_ExposureMode_error.py
ModeCurrent: Timed
Traceback (most recent call last):
  File ".\grab_ExposureMode_error.py", line 42, in <module>
    camera.ExposureMode.SetValue(Timed)
NameError: name 'Timed' is not defined

[grab_ExposureMode_error.txt](https://github.com/basler/pypylon/files/4619818/grab_ExposureMode_error.txt)

thiesmoeller, 2020-05-13:
The argument Timed has to be a string. "Timed"

The error is from python interpreter about undefined variable Timed.

kfuna, 2020-05-13:
Thank you for your prompt reply.
I understand that I need to input it as a string type.
The error has been resolved.

sinasartip, 2020-06-04:
@kfuna 
I think its best if you close the issue since its resolved

closed 2020-06-05 by kfuna