basler/pypylon #235: OnCameraDeviceRemoved listener sometimes doesn't work

yiyunluo07, 2020-05-13:
Hello! I'm using a pua1280-54uc camera from Basler. 

During the test of my program, (in order to find out if it exists some bugs), I tried totally hundreds of times to remove the camera from usb cabel while it is capturing. 

Most of times it worked well with the listener OnCameraDeviceRemoved from pylon.ConfigurationEventHandler, it could call this callback function and I could let the program call an error that the camera is removed.

However, sometimes it didn't call this callback function, when I have removed the camera. When it didn't detect that the camera device is removed, I could not let the program call an error, and when I'm trying to use somewhere else other functions from pypylon, like camera.StopGrabbing() or camera.IsCameraDeviceRemoved(), the program stuck at these functions and cannot move on.

The version of pylon SDK that I use is 5.1, and pypylon 1.4, because they're already outdated, so I also tried to update them and try to test again, if the problem I have still exists. Unfortunately, I still have the same problem.

Is anyone else here has the experience of this problem? It would be nice when someone here can give me some help.

codenics, 2020-05-27:
You may also handle the exceptions thrown when the camera is removed, in this case a pylon.RuntimeException. This would prevent the program from hanging.

thiesmoeller, 2025-12-15:
Closing as answered. When the OnCameraDeviceRemoved callback doesn't fire, handle the exceptions (pylon.RuntimeException) that are thrown when trying to use camera functions after removal. This prevents the program from hanging. The callback may not always fire reliably, so exception handling is the recommended approach for robust camera removal detection.