basler/pypylon #236: looking for a suitable version of PyPylon for Raspberry Pi 4

MevadaRavikumar, 2020-05-13:
Issue 1: My PI version: Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux  || Raspbian 64bits

I have downloaded Pypylon version 1.4.0 using "wget https://github.com/basler/pypylon.../pypylon-1.4.0-cp27-cp27mu-linux_arm7l.whl". It downloads .whl into /home/pi folder. 

When I tried to install it via "Pip3 install pypylon-1.4.0-cp27-cp27mu-linux_arm7l.whl" or "pip install pypylon-1.4.0-cp27-cp27mu-linux_arm7l.whl". It is throwing error like "it is not supported wheel on this platform". 
How can I find out the supported version for my PI? 

Issue 2: Using SDK via ctypes 

      When I tried to run this command: 
       
      #!/usr/bin/env python
      from __future__ import print_function
      import sys
      from IPython.utils._tokenize_py2 import String
      sys.path.append('.')
      import numpy, ctypes
      Pylonlib = "/opt/pylon5/lib64/libpylonc.so"
      from ctypes import cdll
      libc = cdll.LoadLibrary(Pylonlib)
      libc.PylonInitialize()
      #### call any plyon function after you have initialized
      libc.PylonTerminate()

      It is giving me this error: 

          Traceback (most recent call last):
          File "/home/pi/Desktop/cv_ex_1.py", line 6, in <module>
          from IPython.utils._tokenize_py2 import String
          File "/usr/lib/python3/dist-packages/IPython/utils/_tokenize_py2.py", line 46, in <module>
          del x
          NameError: name 'x' is not defined

It would be great if someone can guide me to solve this problem.

thiesmoeller, 2020-05-13:
Some questions
Are you really running aarch64 on rpi4
Do you need python 2.7. it is not supported any more

MevadaRavikumar, 2020-05-14:
Good Morning Thies,

I am using Raspberry pi 4 Model B, 2 GB.

The specification:
Processor: Broadcom BCM2711, quad-core Cortex-A72 (ARM v8) 64-bit SoC @
1.5GHz

I am using Python 2.7. I will start installing Python 3.7.

Regards,
Ravikumar Mevada

MevadaRavikumar, 2020-05-14:
Hi Thies,
Now I am running spyder 3.7.
Tired with this command:

    from __future__ import print_function
    import sys
    import time
    from IPython.utils._tokenize_py2 import String
    sys.path.append('.')
    import numpy, ctypes
    Pylonlib = "/opt/pylon5/lib64/libpylonc.so"
    from ctypes import cdll
    libc = cdll.LoadLibrary(Pylonlib)
    libc.PylonInitialize()
    time.sleep(3)
    libc.PylonTerminate()

Having the same error.
        from IPython.utils._tokenize_py2 import String
        File
"/usr/lib/python3/dist-packages/IPython/utils/_tokenize_py2.py", line 46,
in <module>
           del  x
       NameError: name 'x' is not defined

Regards,
Ravi

thiesmoeller, 2020-05-14:
Your system environment is still unclear:
You write 
4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux 

This is 32bit env not 64bit

In 64bit env you could just install with e.g. python 3.7
`pip3 install https://github.com/basler/pypylon/releases/download/1.5.4/pypylon-1.5.4-cp37-cp37m-linux_aarch64.whl`

Why are you working with ctypes?
pypylon has all integrated

MevadaRavikumar, 2020-05-17:
Hi Thies,

So there is no way to involve PyPylon in 32 bit PI, am I right? Can you
please recommend any 64 bit Raspberry Pi which works fine with Basler
camera? That would be a great help.

Kind Regards,
Ravikumar Mevada

thiesmoeller, 2020-05-17:
You can always compile pypylon.
This will work without any issues on rpi :-)

Follow the readme

thiesmoeller, 2021-02-23:
Beginning with release 1.7.1 we have binary releases on all platforms, that you can install via pip.

update your pip version before by calling `python -m pip install --upgrade pip`

closed 2021-02-23 by thiesmoeller