basler/pypylon #237: MacOS duplicate camera

VladislavSu, 2020-05-16:
`from pypylon import pylon
pylon.__version__
'1.5.4'
from pypylon import pylon
import os 

os.environ['PYLON_CAMEMU'] = "1"
tlFactory = pylon.TlFactory.GetInstance()
devices = tlFactory.EnumerateDevices()

bids =  [p.GetSerialNumber() for p in devices]
print(bids)

['0815-0000', '0815-0000']`

The same issue for real USB camera and GigE camera.
For Linux works fine.

thiesmoeller, 2021-02-23:
can you check with latest update 1.7.1...
---
Beginning with release 1.7.1 we have binary releases on all platforms, that you can install via pip.

update your pip version before by calling `python -m pip install --upgrade pip`

thiesmoeller, 2023-06-14:
Release 2.2.0 with full support of macOS x86_64 and arm64 ( >= macOS 11 ) is out.

Available via pip install or releases page. And also fixes the duplicate camera issue.