basler/pypylon #239: Pypylon API to program for "grabbing an image with hardware trigger

RCC-GIT, 2020-05-19:
Hello, I have problem to make the hardwre trigger working in a loop . 
Tigger evry 500ms to every 1 minutes 
Works OK with Pylonviewer .
Working with ace/aca1920-40gc GIEthernet 
Is it possible to implement a test like 

If camera is triggered :
  Grab image ...

Here is the non working code  with time out error on camera.RetrieveResult

> 
# The parameter MaxNumBuffer can be used to control the count of buffers
# allocated for grabbing. The default value of this parameter is 10.
camera.MaxNumBuffer = 5

# set exposure time
#camera.ExposureTimeRaw.SetValue(100)

# Select the Frame Start trigger 
camera.TriggerSelector.SetValue('FrameStart')
# Acquisition mode
#camera.AcquisitionMode.SetValue('SingleFrame')
camera.AcquisitionMode.SetValue('Continuous')
# Enable triggered image acquisition for the Frame Start trigger 
camera.TriggerMode.SetValue('On') 
# Set the trigger source to Line 1 
camera.TriggerSource.SetValue('Line1')
# Set the trigger activation mode to rising edge 
camera.TriggerActivation.SetValue('RisingEdge')
# Set the delay for the frame start trigger to 10 µs 
camera.TriggerDelayAbs.SetValue(10.0)
# Pixel format
camera.PixelFormat.SetValue('Mono8')

# demarrage acquisition
# =====================================================
camera.StartGrabbing(pylon.GrabStrategy_OneByOne) 
converter = pylon.ImageFormatConverter()

# convertion openCV BGR format 
converter.OutputPixelFormat = pylon.PixelType_BGR8packed 
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
count = 1 
#camera.TriggerSelector.SetValue('AcquisitionStart')
while camera.IsGrabbing():

    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
#   grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_Return)

    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        cv2.namedWindow('title', cv2.WINDOW_NORMAL)
        cv2.imshow('title', img)
        k = cv2.waitKey(1)
        if k == 27:
            break
        
        grabResult.Release()
   
# Releasing the resource  
print("Break")
camera.StopGrabbing()
camera.Close()

cv2.destroyAllWindows()

thiesmoeller, 2020-05-19:
Could you mark your code as code to make it readable

RCC-GIT, 2020-05-19:
```
# The parameter MaxNumBuffer can be used to control the count of buffers
# allocated for grabbing. The default value of this parameter is 10.
camera.MaxNumBuffer = 5

# set exposure time
#camera.ExposureTimeRaw.SetValue(100)

# Select the Frame Start trigger 
camera.TriggerSelector.SetValue('FrameStart')
# Acquisition mode
camera.AcquisitionMode.SetValue('SingleFrame')
# Enable triggered image acquisition for the Frame Start trigger 
camera.TriggerMode.SetValue('On') 
# Set the trigger source to Line 1 
camera.TriggerSource.SetValue('Line1')
# Set the trigger activation mode to rising edge 
camera.TriggerActivation.SetValue('RisingEdge')
# Set the delay for the frame start trigger to 300 µs 
camera.TriggerDelayAbs.SetValue(300.0)
# Pixel format
camera.PixelFormat.SetValue('Mono8')

# demarrage acquisition
# =====================================================
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
converter = pylon.ImageFormatConverter()

# convertion openCV BGR format 
converter.OutputPixelFormat = pylon.PixelType_BGR8packed 
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

count = 1 
while camera.IsGrabbing(): 
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
  #  camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 

    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        print("yes, an image is grabbed successfully")

        cv2.imwrite('save_images/%06d.png'%count, img)
        print("image/%06d.png saved"%count)
        count += 1
        k = cv2.waitKey(1)
        if k == 27:
            break
    
    grabResult.Release()

camera.StopGrabbing()
camera.Close()

```

thiesmoeller, 2020-05-19:
`camera.AcquisitionMode.SetValue('SingleFrame')`

This will stop the camera after one frame.
From the viewpoint of the acquisition engine you have to run in continuous mode
`camera.AcquisitionMode.SetValue('Continuous')`

https://docs.baslerweb.com/acquisition-mode.html

RCC-GIT, 2020-05-19:
**When I have** 
camera.AcquisitionMode.SetValue('SingleFrame')
**The result is** 
xxx   line 77, in <module>
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

TimeoutException: Grab timed out. The acquisition is not started. In 'SingleFrame' acquisition mode the acquisition has to be started for every frame, see also GrabOne(). : TimeoutException thrown (file 'instantcameraimpl.h', line 1051)

**When I have** 
camera.AcquisitionMode.SetValue('Continuous')
**The result is the same**
line 78, in <module>
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

  File "C:\Users\HRC\Anaconda3\lib\site-packages\pypylon\pylon.py", line 3458, in RetrieveResult
    return _pylon.InstantCamera_RetrieveResult(self, *args)

TimeoutException: Grab timed out. The acquisition is not started. : TimeoutException thrown (file 'instantcameraimpl.h', line 1056)

**The trigger is working for 5000 xxs and I am able to grab an image . Atfer that I have the time out exception**

thiesmoeller, 2020-05-19:
What do you mean by "5000 xxs"?

RCC-GIT, 2020-05-19:
5000ms =5s  this is the time windows on which I am able to grab an image 
If I test at 50000ms =50s I have more time. but I want to able to grab for ever and wait for the trigger at any time .

thiesmoeller, 2020-05-19:
In this case best to use the OnImageGrabbed callback as described in the grabusinggrabloopthread.py sample

The background thread waits forever until an image is grabbed and then calls the handler where you can save your image

RCC-GIT, 2020-05-20:
Thank you for your answer . I need some clarifications 
This grabusinggrabloopthread.py is for software trigger and Key action for image event handler

So in line :
`camera.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(),pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)`
**pylon.SoftwareTriggerConfiguration()** must be replace what ?
Same for :
`camera.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)`
**pylon.GrabStrategy_OneByOne**  is this correct ?

Also  :
```
while True:
            time.sleep(0.05)
            key = getkey()
            print(key)
            if (key == 't' or key == 'T'):
                # Execute the software trigger. Wait up to 100 ms for the camera to be ready for trigger.
                if camera.WaitForFrameTriggerReady(100, pylon.TimeoutHandling_ThrowException):
                    camera.ExecuteSoftwareTrigger();
            if (key == 'e') or (key == 'E'):
                break
```
How do I replace Key action t or T by Hardware Trigger action in This While loop ? Maybe this can work as the while loop will get an interruption from the image handler   ?
```
while True:
            time.sleep(0.05)
            key = getkey()
            if (key == 'e') or (key == 'E'):
                break
```

And  I suppose i will grab the image in the OnImageGrabbed subroutine ? Please confirm.

By the end my target is to have one unique trigger and 4 cameras , hope this program will work fo rmore then one camera .

Charli-py, 2020-09-18:
Hello, @thiesmoeller 
I am trying to capture image of a moving object with a line scan camera. But iam getting grabtimeout Error.
Can you please help iam new to this application

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

camera.Open()

print("DeviceClass: ", camera.GetDeviceInfo().GetDeviceClass())
print("DeviceFactory: ", camera.GetDeviceInfo().GetDeviceFactory())
print("ModelName: ", camera.GetDeviceInfo().GetModelName())

Hardware_Trigger = False

if Hardware_Trigger:
# reset registration
 camera.RegisterConfiguration(pylon.ConfigurationEventHandler(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)

# The parameter MaxNumBuffer can be used to control the count of buffers
# allocated for grabbing. The default value of this parameter is 10.
camera.MaxNumBuffer = 10

# set exposure time
camera.ExposureTimeRaw.SetValue(100)

# Select the Frame Start trigger 
camera.TriggerSelector.SetValue('FrameStart')
# Acquisition mode
camera.AcquisitionMode.SetValue('Continuous')
# Enable triggered image acquisition for the Frame Start trigger 
camera.TriggerMode.SetValue('On') 
# Set the trigger source to Line 1 
camera.TriggerSource.SetValue('Line1')
# Set the trigger activation mode to rising edge 
camera.TriggerActivation.SetValue('RisingEdge')
# Set the delay for the frame start trigger to 300 µs 
camera.TriggerDelayAbs.SetValue(10.0)
# Pixel format
camera.PixelFormat.SetValue('Mono8')

camera.StartGrabbing(pylon.GrabStrategy_OneByOne)
converter = pylon.ImageFormatConverter()

converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

count = 1
while camera.IsGrabbing():
    
    
    
#     camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        
    if grabResult.GrabSucceeded():
    # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        print("yes, an image is grabbed successfully")

        cv2.imwrite('D:\Print\save_images/%06d.png'%count, img)
        print("image/%06d.png saved"%count)
        count += 1
        k = cv2.waitKey(1)
        
        
    grabResult.Release()

camera.StopGrabbing()
camera.Close()

thiesmoeller, 2020-09-18:
Please edit your post according to https://guides.github.com/features/mastering-markdown/# for code formatting to make it readable.

Can you grab images in pylon viewer in this configuration?

Charli-py, 2020-09-23:
Sorry for the late reply.
Yes iam able to grab the image in pylon viewer in continuous mode with this configuration
 

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

camera.Open()

print("DeviceClass: ", camera.GetDeviceInfo().GetDeviceClass())
print("DeviceFactory: ", camera.GetDeviceInfo().GetDeviceFactory())
print("ModelName: ", camera.GetDeviceInfo().GetModelName())

Hardware_Trigger = False

if Hardware_Trigger:

 camera.RegisterConfiguration(pylon.ConfigurationEventHandler(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)

camera.MaxNumBuffer = 10

camera.ExposureTimeRaw.SetValue(100)

 
camera.TriggerSelector.SetValue('FrameStart')
camera.AcquisitionMode.SetValue('Continuous')
 
camera.TriggerMode.SetValue('On') 
 
camera.TriggerSource.SetValue('Line1')
 
camera.TriggerActivation.SetValue('RisingEdge')
 
camera.TriggerDelayAbs.SetValue(10.0)
camera.PixelFormat.SetValue('Mono8')

camera.StartGrabbing(pylon.GrabStrategy_OneByOne)
converter = pylon.ImageFormatConverter()

converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

count = 1
while camera.IsGrabbing():
    
    
    

    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        
    if grabResult.GrabSucceeded():
    
        image = converter.Convert(grabResult)
        img = image.GetArray()
        print("yes, an image is grabbed successfully")

        cv2.imwrite('D:\Print\save_images/%06d.png'%count, img)
        print("image/%06d.png saved"%count)
        count += 1
        k = cv2.waitKey(1)
        
        
    grabResult.Release()

camera.StopGrabbing()
camera.Close()

thiesmoeller, 2020-09-23:
what so you want to achieve?
You disabled the path 'hardware trigger' .. the operation that you skipped is correct to reset any default configurations..

In the grab loop you wait for a key press.. why ... 

You convert the mono8 image to rgb....png supports mono video data....

And to repeat the question.. can you successfully run the _hw triggered_ usecase in pylon viewer?

thiesmoeller, 2021-02-23:
@Charli-py : did you made any progress ?

simonsauss, 2021-04-21:
Hello,

I have a basler camera and I'm trying to set triggers with the grabusinggrabloopthread.py script. For example, I would like to record 10 images every 1 second then 50 images every 20 seconds etc... Does anyone know how I could do this? I can't find the commands to do this. For information I have an exposure time of 66 000 us...

Thanks

ridvanozdemir, 2021-07-29:
grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
how about if I don't want to use any timeout parameter

thiesmoeller, 2021-12-16:
Please check our companion repository https://github.com/basler/pypylon-samples where HW trigger and much more is explained

gt8mar, 2022-03-15:
Is there a way to stream video from one camera, then have a waitkey that triggers saving a series of images from one camera and a second HW triggered camera?

I'm imagining setting a key like "r" to mean record. Then stream video without saving until I press "r", at which point it begins saving. 

I tried using input() but that stops the loop where I stream the video.

thiesmoeller, 2022-03-16:
Hi @gt8mar ,
Please move your question to a new issue.

thiesmoeller, 2025-12-15:
Closing as answered. For hardware trigger with continuous waiting, use the OnImageGrabbed callback approach as shown in grabusinggrabloopthread.py sample. Set AcquisitionMode to 'Continuous', configure hardware trigger, then use StartGrabbing with GrabLoop_ProvidedByInstantCamera. The background thread waits for triggers and calls your handler. See https://github.com/basler/pypylon-samples for complete hardware trigger examples.

closed 2025-12-15 by thiesmoeller